iiab/iiab #938: Kolibri reinstall fails on RPi

holta, 2018-07-20:
@arky is this presumably an intermittent thing?  While the original `./iiab-install` worked, output of `./iiab-install --reinstall` is below:

> TASK [kolibri : Install kolibri using pip on all OS's] *************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/bin/pip2 install -U --no-cache-dir kolibri", "msg": "\n:stderr: Exception:\nTraceback (most recent call last):\n  File \"/usr/lib/python2.7/dist-packages/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/usr/lib/python2.7/dist-packages/pip/commands/install.py\", line 342, in run\n    requirement_set.prepare_files(finder)\n  File \"/usr/lib/python2.7/dist-packages/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/usr/lib/python2.7/dist-packages/pip/req/req_set.py\", line 487, in _prepare_file\n    req_to_install, finder)\n  File \"/usr/lib/python2.7/dist-packages/pip/req/req_set.py\", line 428, in _check_skip_installed\n    req_to_install, upgrade_allowed)\n  File \"/usr/lib/python2.7/dist-packages/pip/index.py\", line 465, in find_requirement\n    all_candidates = self.find_all_candidates(req.name)\n  File \"/usr/lib/python2.7/dist-packages/pip/index.py\", line 423, in find_all_candidates\n    for page in self._get_pages(url_locations, project_name):\n  File \"/usr/lib/python2.7/dist-packages/pip/index.py\", line 568, in _get_pages\n    page = self._get_page(location)\n  File \"/usr/lib/python2.7/dist-packages/pip/index.py\", line 683, in _get_page\n    return HTMLPage.get_page(link, session=self.session)\n  File \"/usr/lib/python2.7/dist-packages/pip/index.py\", line 792, in get_page\n    \"Cache-Control\": \"max-age=600\",\n  File \"/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 501, in get\n    return self.request('GET', url, **kwargs)\n  File \"/usr/lib/python2.7/dist-packages/pip/download.py\", line 386, in request\n    return super(PipSession, self).request(method, url, *args, **kwargs)\n  File \"/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 488, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 609, in send\n    r = adapter.send(request, **kwargs)\n  File \"/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py\", line 423, in send\n    timeout=timeout\n  File \"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 643, in urlopen\n    _stacktrace=sys.exc_info()[2])\n  File \"/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py\", line 315, in increment\n    total -= 1\nTypeError: unsupported operand type(s) for -=: 'Retry' and 'int'\n"}
        to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry
>
> PLAY RECAP *********************************************************************
> 127.0.0.1                  : ok=292  changed=64   unreachable=0    failed=1

Clarif 1: `./runrole kolibri` worked, right after the above failure.  Let's try hard to understand & keep an eye on this (reducing the frequency of such failure modes) as new contributors deserve a very reliable IIAB installation path, to keep focus and build confidence!

Clarif 2: Earlier today, IIAB 6.6/master was installed on the latest Raspbian Lite 2018-06-27 (updated with `apt update; apt -y dist-upgrade; reboot`) on RPi 3 B+ using install script http://download.iiab.io/6.6/install.txt

arky, 2018-07-21:
This error could be caused by network connectivity issues (connection timeouts or need to set http_proxy environment variable).

Another issues is conflicts between ['urllib3'](
https://pypi.org/project/urllib3/) and 'requests' module. You can resolve it by updating request module and removing urllib3. 
```
pip install --upgrade pip
pip install --upgrade request 
pip uninstall urllib3

```
Please let me know if this resolves the issue. 

@holta  It is not related to #920

holta, 2018-08-08:
Using `pip` to install the latest Kolibi worked on 6+ different machines recently (RPi 3, RPi 3 B+, Ubuntu 16.04, Ubuntu 18.04) upgrading from 0.10.0 to 0.10.1

So I'm going to close this tkt &mdash; but plz re-open as nec, if you encounter the same.

closed 2018-08-08 by holta