iiab/iiab #941: http://10.0.0.x/moodle not working

holta, 2018-07-21:
@kananigit and others would like Moodle 3.5.1 LTS to work across "campus", any idea what Moodle settings allow it to be available beyond hard-coded/primary IP like internal Wi-Fi hotspot's http://172.18.96.1/moodle ?

Related:
#923 "Failing URL's—parent tkt to consolidate diverse issues"

kananigit, 2018-07-21:
So this is my encounter

I am trying to access moodle from iiab in an ubuntu machine in my network. The IP address is 10.0.0156, from a different computer am able to access 10.0.0.156/home   but when i try to click on the moodle menu item the url defaults to box.lan/moodle and that page is just not found.

Am however able to access nextcloud because it defaults to 10.0.0.156/nexcloud.

how can we make moodle resolve correctly like other apps. KA lite works just fine too on 10.0.0.156:8008

holta, 2018-07-21:
@kananigit if you want a quick+dirty solution it looks like you can hard-coded whatever Moodle URL you want into https://github.com/iiab/iiab/blob/master/roles/moodle/templates/moodle_installer#L10

Of course that means modifying Line 10 of /opt/iiab/iiab/roles/moodle/templates/moodle_installer within IIAB.

If possible do this prior to the 1st run of "cd /opt/iiab/iiab" then "./iiab-install" (i.e. the core of IIAB's install process).

If however IIAB & Moodle are already installed, forcing Moodle to reinstall takes a bit more than just "cd /opt/iiab/iiab" then "./runrole moodle" — as you can work through using http://FAQ.IIAB.IO #23 ("Can I upgrade or reinstall server apps?")  Anyway getting to know  Moodle and how to force it to partially-or-completely reinstall with a truly clean slate is something you need to learn regardless (if you will be promoting Moodle 3.5.1 LTS as part of your IIAB product!)

Quoting from FAQ.IIAB.IO #23:

"[delete] /opt/iiab/moodle/config-dist.php [prior to "./runrole moodle" to try to force Moodle to upgrade or reinstall] (Moodle version stored in /opt/iiab/moodle/version.php — major upgrades may require deleting /opt/iiab/moodle and /library/moodle, dropping the database using "dropdb -U postgres moodle", and worst case upgrading PostgreSQL [...] )"

holta, 2018-07-21:
Or perhaps you'll find an even cleaner solution to modify Moodle's "mandatory" --wwwroot=URL without having to (re)run Moodle's install.php (as mentioned in Line 10 of moodle_installer in IIAB's Ansible playbook above).

jvonau, 2018-07-26:
Could this might be solved by using localhost for --wwwroot and a proxypass definition in the apache config file?

holta, 2018-08-05:
@kanagit please provide us an update in coming days (what you've tried, what works, what doesn't!)

(As IIAB 6.6 Release Candidate 1 is about to be released and others need to work in concert here with you, Thanks!)

kananigit, 2018-08-05:
Definitely . will write something up by tomorrow.

kananigit, 2018-08-06:
Changing       /opt/iiab/moodle/config.php  line $CFG->wwroot = 'http://10.0.0.156/moodle';     for my case worked

holta, 2018-08-06:
> Changing /opt/iiab/moodle/config.php line $CFG->wwroot = 'http://10.0.0.156/moodle'; for my case worked

Is this generalizable to all IP addresses, similar to @arky's PR #980 for http://box/kiwix ?

(I'd like to clean up all these URL's, so http://box/moodle works for almost all, if indeed that's achievable!)

holta, 2019-10-21:
This is not much a bug with IIAB, than it is a fact of life with the way Moodle works.

Please re-open if you'd like to contribute to a programmatic workaround or solution.

closed 2019-10-21 by holta