iiab/iiab #942: MongoDB's unit file "mongod --repair" blocks on "dirty journal files" [Sugarizer]

holta, 2018-07-21:
Running `./runrole mongodb` or `./runrole sugarizer` fail as follows:

> 2018-07-21 20:13:58,729 p=6529 u=root |  TASK [mongodb : Populate & position /etc/mongod.conf, /etc/systemd/system/mongodb.service] ***
2018-07-21 20:14:00,426 p=6529 u=root |  ok: [127.0.0.1] => (item={u'dest': u'/etc/systemd/system/mongodb.service', u'src': u'mongodb.service.j2'})
2018-07-21 20:14:01,909 p=6529 u=root |  ok: [127.0.0.1] => (item={u'dest': u'/etc/mongod.conf', u'src': u'mongod.conf.j2'})
2018-07-21 20:14:02,113 p=6529 u=root |  TASK [mongodb : Restart service if enabled] ************************************
2018-07-21 20:14:04,084 p=6529 u=root |  failed: [127.0.0.1] (item={u'name': u'mongodb'}) => {"changed": false, "item": {"name": "mongodb"}, "msg": "Unable to start ser
vice mongodb: Job for mongodb.service failed because the control process exited with error code.\nSee \"systemctl status mongodb.service\" and \"journalctl -xe\" for de
tails.\n"}
2018-07-21 20:14:04,092 p=6529 u=root |         to retry, use: --limit @/opt/iiab/iiab/run-one-role.retry
>
> 2018-07-21 20:14:04,093 p=6529 u=root |  PLAY RECAP
> *********************************************************************
> 2018-07-21 20:14:04,094 p=6529 u=root |  127.0.0.1                  : ok=25   changed=5    unreachable=0    failed=1

This is a result of @jvonau's very noble attempt (#919) to run "mongod --repair" every time systemd starts up MongoDB.

The reason this fails is shown by the output of `sudo -u mongodb /usr/bin/mongod --repair --dbpath /library/dbdata/mongodb` :

>...
> You specified --repair but there are dirty journal files. Please
> restart without --repair to allow the journal files to be replayed.
> If you wish to repair all databases, please shutdown cleanly and
> run with --repair again.
> **************
> Sat Jul 21 21:51:20.023 [initandlisten] exception in initAndListen: 12596 old lock file, terminating
> Sat Jul 21 21:51:20.023 dbexit:
> ...

@jvonau or @llaske do you have suggestions...ideally preserving `mongod --repair` but if so only running it...when it truly works and is helpful ?

llaske, 2018-07-21:
Never seen this message. BTW,  `mongod --repair` should be launch before `mongod`. This error message let me think that both are launched at the same time.
It's very important the you wait for the end of `mongod --repair` before launching the `mongod` command. Are you sure it's correct ?

jvonau, 2018-07-22:
This looks like a reinstall where mongodb is already running or has been started as part of the 'normal' apt routine given there is a '[initandlisten] exception in initAndListen: 12596 old lock file, terminating'. Think what is missing is a systemctl deamon-reload when the new unit file is placed into position, think 'service' should be replaced with 'systemd' and add the reload option.

holta, 2018-07-22:
PRs #945 and #946 (working off @jvonau suggestions above) have been merged so that TK and others can now please try to break MongoDB & Sugarizer.

In repeatable/reproducible ways~

Thanks You!

holta, 2018-07-23:
If PR #957 works well for TK/others this tkt should be close.

holta, 2018-07-27:
TK's been unable to reproduce this in for many days now.

Am closing for now.  Re-open as nec, if it recurs!

closed 2018-07-27 by holta