iiab/iiab #953: 'apt update' failing frequently on RPi over past 3 days (archive.raspberrypi.org flaw?)

holta, 2018-07-22:
Do we have any more of working around this frequent error message that has bedeviled IIAB installs (and any execution of  `apt update`) over the past 3 days?

> E: The repository 'http://archive.raspberrypi.org/debian stretch Release' **does no longer have a Release file**.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

holta, 2018-07-23:
Tangentially related...to intermittent download/install failures:

#938 "Kolibri pip (re)install intermittently fails on RPi"

holta, 2018-07-23:
This problem is getting significantly worse 5 days later, e.g. both yesterday and today.

It is ridiculous that people have to run commands like "apt update" and "./runrole sugarizer" 5 times in a how just to get past the `The repository 'http://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.` failure.

FWIW this happens particularly frequently in Sugarizer's playbook at this point: `Set up Node.js 8.x apt sources (debuntu, but avoid ubuntu-18)`

(But I don't see any alternative, until these upstream download sites clean up their act?)

arky, 2018-07-23:
Here is solution for this issue:  Look at the nearest mirror and change the /etc/apt/sources.list file 
https://www.raspbian.org/RaspbianMirrors

deb http://mirror.nus.edu.sg/raspbian/raspbian stretch main contrib non-free rpi
#deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi

https://linuxconfig.org/how-to-find-a-fastest-debian-linux-mirror-for-your-etc-apt-sources-list

holta, 2018-08-05:
I've been unable to reproduce this over the past week, so am moving this issue to [IIAB 6.7](https://github.com/iiab/iiab/milestone/4)

Likely this will flare up again when archive.raspberrypi.org 's mirroring infra cracks under pressure once again in coming months?

holta, 2018-09-13:
This has not been a problem since July, but let's keep an eye on it when Raspbian publishes their latest OS at https://www.raspberrypi.org/downloads/raspbian/ (in coming days presumably?)

holta, 2018-10-08:
Not a problem since July.

closed 2018-10-08 by holta