iiab/iiab #958: Non-ASCII Characters in Install.txt Bash Script

arky, 2018-07-23:
Please kindly avoid using non-ASCII characters in IIAB install script. These characters can not be rendered in a standard terminal, instead shows up like square blobs 'aka tofu'. 

```
    echo -e "\n\n  EXISTING /etc/iiab/local_vars.yml will be used to install Internet-in-a-Box\n"
    echo -e "   🚂 🚃 🚄 🚅 🚆 🚇 🚈 🚉 🚊 🚋 🚌 🚍 🚎 🚏 🚐 🚑 🚒 🚚 🚛 🚜 🚞 🚟 🚠 🚡 🚲\n

```
It is a non-standard practice that might confuse some new users. They might think something wrong with the install scrip.

holta, 2018-07-23:
So far the females and newbies I've shown http://download.iiab.io/6.6/install.txt to, greatly appreciate the friendly iconography.

But if others go bananas when these modern Unicode characters do not appear in some terminals we can consider removing them.

Certainly we can all agree that a friendly installer is critically important.  As our installer improves over time, possibly it can auto-detect which terminals have modern Unicode characters installed and which do not.

arky, 2018-07-23:
The default en-GB terminal of Raspbian Lite didn't support/show these characters.

holta, 2018-07-29:
> The default en-GB terminal of Raspbian Lite didn't support/show these characters.

When I try the same thing (default locale en_GB.UTF-8 on Raspbian Lite) these Unicode characters show up perfectly.

They also work when I use raspi-config to set the "default locale" to en_GB.UTF-8, en_US.UTF-8, es_ES.UTF-8 and es_MX.UTF-8.

Finally I also tried changing the "default locale" to C.UTF-8 and also to None (which appears as "POSIX" as I'm pasting in below.)

Also on other OS's like Ubuntu these Unicode characters appear properly.

(@arky if you have ideas: what else should I be trying?  So far I'm completely unable to make it fail...FWIW I'm traveling so using ssh rather than an attached video/monitor at this time...if this makes any difference?)
```
root@box:~# more /etc/default/locale
#  File generated by update-locale
#LANG=en_GB.UTF-8

root@box:~# locale
LANG=
LANGUAGE=
LC_CTYPE="POSIX"
LC_NUMERIC="POSIX"
LC_TIME="POSIX"
LC_COLLATE="POSIX"
LC_MONETARY="POSIX"
LC_MESSAGES="POSIX"
LC_PAPER="POSIX"
LC_NAME="POSIX"
LC_ADDRESS="POSIX"
LC_TELEPHONE="POSIX"
LC_MEASUREMENT="POSIX"
LC_IDENTIFICATION="POSIX"
LC_ALL=
```

holta, 2018-08-22:
So far I haven't been able to reproduce this.

If ambiguities remain, let's resolve this as part of IIAB 6.7

closed 2019-10-01 by holta