iiab/iiab #963: Move logic from web curl script to iiab-install or factory

jvonau, 2018-07-24:
Move the bulk of [web curl script](http://download.iiab.io/6.6/install.txt) to iiab-install to avoid duplication and provide easier maintenance or move install.txt to factory and point the url to github.

holta, 2018-07-24:
Related:
PRs #931 #964 #965

tim-moody, 2018-07-24:
I'd like to see a separate script for setup: clone repos, create initial dir, etc. before the actual install starts.

jvonau, 2018-07-25:
[Move to github](https://github.com/iiab/iiab-factory/pull/33/files)

holta, 2018-10-01:
Refs:
- https://github.com/iiab/iiab-factory/pull/33
- #1160 "Proactively discourage pi/raspberry password compromises"
- http://download.iiab.io/6.7/install.txt (major UX improvements over the past 2 months, so newbie implementers don't forget to reboot into new kernel, ETC)

closed 2018-12-10 by holta