iiab/iiab #972: Validate Nextcloud 13.0.6 on "2018-08-30"

holta, 2018-07-27:
Schedule for now: https://github.com/nextcloud/server/wiki/Maintenance-and-Release-Schedule

floydianslips, 2018-08-30:
Upgrading to Nextcloud 13.0.6 creates an integrity issue with unneeded files. This script will delete the files in question and remove the error.
downloads.joshuadennis.ca/ncug.sh

holta, 2018-08-30:
Nextcloud 13.0.6 (https://nextcloud.com/changelog/) tested on Ubuntu 18.04 and Raspbian.

@floydianslips suggested script http://keys.joshuadennis.ca/ncug.sh to tamp down some of the red security warnings that "always" appears at http://box/nextcloud/index.php/settings/admin:

> * Some files have not passed the integrity check. Further information on how to resolve this issue can be found in the documentation. (List of invalid files… / Rescan…)
>
> * Your data directory and files are probably accessible from the Internet. The .htaccess file is not working. It is strongly recommended that you configure your web server so that the data directory is no longer accessible, or move the data directory outside the web server document root.
>
> * Accessing site insecurely via HTTP. You are strongly adviced to set up your server to require HTTPS instead, as described in the security tips.

He clarifies:

> the htaccess is always [there]
> this cures the top integrity issue

> I have fixed the other two previously and can try again for a permanent solution
> I have very little time at the moment unfortunately

holta, 2018-08-30:
Let's move this discussion to IIAB 6.7 tkt #930 ("https://github.com/iiab/iiab/issues/930")

holta, 2018-08-30:
Recap: 13.0.6 has been tested and seems to work well on Raspbian and Ubuntu 18.04 for IIAB 6.6's imminent final release.

closed 2018-08-30 by holta