iiab/iiab #974: Sugarizer 1.0.1 Release Polish

holta, 2018-07-27:
Highest Priority Sugarizer 1.0 -> 1.0.1 issues from https://github.com/iiab/iiab/pull/888#issuecomment-404370082 (and https://github.com/iiab/iiab/issues/798#issuecomment-405609004) summarized here:

- [ ] #898 Fix http://box/sugarizer (worked with Sugarizer 0.7, 0.8, 0.9)
- [ ] #899 Enable (or mitigate!) Debian 9 rsyslog service
- [ ] #915 Yank power from MongoDB and see if Sugarizer 1.0 works on next boot(s) 
- [ ] #921 Optimize Sugarizer disk footprint/speed by "50%" using grunt

jvonau, 2020-09-26:
Review sub issue and close stale ones.

jvonau, 2021-07-21:
Works in RPi Zero W? Y/N is a FAQ and docker is out close.

holta, 2021-07-21:
I don't know if Sugarizer works on RPi Zero W.

jvonau, 2021-07-21:
it's subject to the nodejs version notes about arm6l and manual steps/rollback, or just own up and say UN-supported on arm6l.

jvonau, 2021-08-29:
#2739

jvonau, 2022-04-05:
#915 is closed, the rest is stale.

holta, 2022-04-05:
Good enough, 4 years later (as lot of work has gone into this!!)

closed 2022-04-05 by holta