iiab/iiab #977: Kolibri 0.10.0+ Release Polish

holta, 2018-07-27:
Highest priorities to bring Kolibri into the hands of mainstream teachers &mdash; building off #841:
- [ ] #884 Automate torrenting+installing of KA Lite videos in 7 languages
- [ ] #913 http://box/kolibri should work! (not just http://box:8009)
- [ ] #920 Is 500MB swap file best on RPi? [pip cache memory issues; installing Kolibri, KA Lite & OSM?]
- [ ] #938 Kolibri pip (re)install intermittently fails on RPi
- [ ] #922 Verify Kolibri 0.10.0 -> 0.10.1 or 0.11.x ?
- [ ] Understand why Kolibri appears slow on RPi?
- [ ] _Friendly guidance on integrating Other Content (@ivanistheone will record the video ;)_

arky, 2018-09-05:
Requested kolibri developers to release development pip packages to enable testing. 

https://github.com/learningequality/kolibri/issues/4218

holta, 2019-03-06:
Please follow up on our new Kolibri checklist here:

https://github.com/iiab/iiab/issues/1545

closed 2019-03-06 by holta