iiab/iiab #981: [FAQ] How to add zim files to Kiwix server

arky, 2018-08-06:
@holta Perhaps we need to document this somewhere. 

### How to add new .zim files to Kiwix Server?
- Download a zim file to /library/zims/content
```
$ cd /library/zims/content/
$ sudo wget https://download.kiwix.org/zim/wikipedia/wikipedia_km_all_nopic_2018-04.zim 
```
- Run iiab-make-kiwix-lib to rebuild the Kiwix index.
`$ sudo /usr/bin/iiab-make-kiwix-lib`
- Go to Kiwix home page box/kiwix

holta, 2018-08-06:
Links to these 2 places should also be considered:
- https://github.com/iiab/iiab/wiki/IIAB-Installation#zim-files
- https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst``

holta, 2018-08-06:
Done- please all help refine "How do I add ZIM files, like Wikipedia?" which is currently http://FAQ.IIAB.IO Question 27.

closed 2018-08-06 by holta