iiab/iiab #987: Add backports Repo to Raspbian / Debian Distributions

arky, 2018-08-08:
Unlike Ubuntu, Debian release cycle follows a measured pace. Some of the newer packages are added via [Debian Backports](https://backports.debian.org/). 

```
deb http://ftp.debian.org/debian stretch-backports main
apt-get -t stretch-backports install "package"
```
You can see the full list of packages that part of Debian Backport [here](https://backports.debian.org/Packages/) and those in new queue [here](https://ftp-master.debian.org/backports-new.html)

holta, 2018-08-08:
Is this safe to add to Raspbian?

(Just wondering as we've had stability problems with Raspbian even without this...)

holta, 2018-08-15:
Question for all: what packages does IIAB really need from https://packages.debian.org/stretch-backports/ (if any) on Raspbian especially?

holta, 2018-08-15:
Thanks @arky for finding those.

Is a PR for debuntu advisable, including something like the following?
```
echo "deb http://raspbian.raspberrypi.org/raspbian/ testing main" > /etc/apt/sources.list
```
(Or should that be part of a specific backport/PR when its need is confirmed needed?)

arky, 2018-08-15:
@holta Please kindly assign this bug to me. Am going to test on Raspbian Stretch tomorrow. 

Where would one patch this for Raspbian Stretch Lite / Desktop?

holta, 2018-08-15:
> Where would one patch this for Raspbian Stretch Lite / Desktop?

I don't know so am cc'ing @jvonau, @tim-moody, @georgejhunt.  Personally I'd recommend isolating the change as follows:
```
echo "deb http://ftp.debian.org/debian stretch-backports main" > /etc/apt/sources.list.d/stretch-backports.list
```
Or perhaps better:
```
echo "deb http://ftp.debian.org/debian stretch-backports main" > /etc/apt/sources.list.d/iiab-stretch-backports.list
```

arky, 2018-08-29:
Reading through the code, it seem that ']2-common](https://github.com/iiab/iiab/tree/50b792869d65114daccfc77ab290b16295e207a2/roles/2-common)' roles is the idea place for this. Can someone confirm this?

Also what is right variables to detect 'Raspbian' OS version?

holta, 2018-08-29:
> what is right variables to detect 'Raspbian' OS version?

Are you asking for the `is_rpi` variable in https://github.com/iiab/iiab/blob/master/vars/raspbian-9.yml#L1 ?

arky, 2018-08-29:
Perhaps I think we need a new variable 'is_rpi_9' equivalent to other distro's like ubuntu/debian in '[raspbian-9](
https://github.com/iiab/iiab/blob/50b792869d65114daccfc77ab290b16295e207a2/vars/raspbian-9.yml)'

https://github.com/iiab/iiab/blob/50b792869d65114daccfc77ab290b16295e207a2/vars/ubuntu-18.yml#L3

holta, 2018-08-30:
> Perhaps we need a new variable 'is_rpi_9' equivalent to other distro's like ubuntu/debian in '/var/raspbian-9'

@arky please go ahead.

1) There is some debate that this _should_ really be called is_raspbian_9 instea of is_rpi_9 so that x86_64 and i686 architectures are handled properly.

2) Which you choose, please set the default to False in https://github.com/iiab/iiab/blob/master/vars/default_vars.yml

holta, 2018-08-30:
@jvonau asks if you can restrict backports to only those things required for cockpit (#1037) ?

arky, 2018-08-30:
@holta  Thanks, I will think of sensible variable name and set its value as advised. 

@jvonau  Backport packages are apt pinned to prevent problems.  See  https://github.com/iiab/iiab/issues/987#issuecomment-411662832

georgejhunt, 2018-08-30:
> prevent problems. See #987 (comment)

Perhaps we already have the variable you need:

https://github.com/iiab/iiab/blob/master/vars/raspbian-9.yml  -- line 1

jvonau, 2020-04-29:
stale

closed 2020-04-29 by holta