iiab/iiab #991: install.txt on Raspberry PI  Stretch 9.4 booting from USB failing

shantibhardwa, 2018-08-08:
### Expected behavior and actual behavior.

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.

shantibhardwa, 2018-08-08:
I am trying to do iiab build on Stretch 9.4 on PI3B Plus which is booting from USB  Flash drive using the OTP bit set ( i.e. no SD card in the PI). tune2fs gives an error saying No such file or directory while trying to open /dev/mmcblk0p2.  Couldn't find valid filesystem superblock. As there is no SD card the root partition with the file system  is on the USB  and  /dev/mmcblk0p2 does not exist..

closed 2018-08-08 by shantibhardwa

holta, 2018-08-08:
@shantibhardwa what was the fix if possible, for the benefit of others?