iiab/iiab #992: http://10.0.0.x/wordpress/  not displaying correctly  & http://10.0.0.x/wordpress/wp-admin/ not working

kananigit, 2018-08-08:
When trying to access wordpress over the network, the display is totally distorted and broken
Here is a snapshot from trying to access http://10.0.0.x/wordpress/ 
[image]

The styling is not picked up for some reason. 

On the other hand,
http://10.0.0.x/wordpress/wp-admin/  is NOT working. 
Instead it resolves to http://127.0.0.1/wordpress/wp-admin/  
which won't work since its trying to fetch wordpress from my current machine over the network instead of the actual server.

I have a feeling its a URL issue like the one we had with Moodle [#941](https://github.com/iiab/iiab/issues/941). Still trying to figure it out.

### Specifications like the version of the IIAB, operating system version, or hardware details.
I am using Ubuntu 18.04 as a server that has IIAb 6.5. I also tried using  RPI 3B containin =g IIAB 6.5 but had the same issue.

kananigit, 2018-08-09:
###What i have tried
Going to Admin Panel > Settings > General and replace localhost with your 10.0.0.x for WordPress Address (URL) and Site Address (URL)

Did the same thing also by setting the following constants in wp-config.php in wordpress
define('WP_SITEURL', 'http://10.0.0.x/wordpress');
define('WP_HOME', 'http://10.0.0.x/wordpress');

This  retained **some** CSS when viewing the wordpress site from another computer but it was still not very good and the  display in a phone was really ugly.

http://10.0.0.x/wordpress/wp-admin  resolved correctly too this time but the styling was not as good

holta, 2018-08-09:
@kananigit it's critical that the Very First Use of WordPress occur from a browser that is using an IP address in conformance with the IP range of the actual users (10.0.0.x in this case?!)

Can you wipe WordPress from your system completely[*] then try registering WordPress from such a http://10.0.0.x IP address?

[*] Summary Instructions on wiping /library/wordpress and dropping its MySQL "iiab_wp" database at http://FAQ.IIAB.IO QUESTION 23 ("Can I upgrade or reinstall server apps?")

Also Related: #201 "Automate re-homing of http://<HOSTNAME>/wordpress? (instead of RELOCATE flag)"

kananigit, 2018-08-09:
@holta  accessing wordpress on the server itself or from a device connected to the server's WIFI hotspot works flawlessly. The issue starts when i connect the server via Ethernet to my home router then try accessing it from another computer on the home network.

kananigit, 2018-08-09:
> @kananigit <https://github.com/kananigit> I'm closing this ticket based
> on the understanding that 1 installation of WordPress is not capable of
> supporting 2 different URL's.
>
> If WordPress fixes this situation in future, that will be very positive — *if
> you find any upstream design discussions within WordPress forums (etc)
> definitely please post here* — i.e. if progress becomes possible towards
> solving this longstanding WordPress limitation.

That's fine. Will be on the lookout for any updates from wordpress

holta, 2018-08-09:
@tim-moody recommends https://wordpress.org/plugins/multiple-domain/

Am re-opening this ticket in case someone wants to attack this for [IIAB 6.7](https://github.com/iiab/iiab/milestone/4).

holta, 2019-10-01:
FYI this tkt is almost identical to:

#201 Automate re-homing of http://<HOSTNAME>/wordpress? (instead of RELOCATE flag)
#551 How to mitigate campuswide failures for Nextcloud, WordPress, Moodle, Elgg etc?

jvonau, 2021-08-29:
#2739