iiab/iiab #993: Access Calibre Server without login

arky, 2018-08-09:
I don't if this is possible with Calibre. But we could display some example books without the need to login. This is useful for mobile library public access system and technical demos. 

Testcase: 
1. Open box/ebooks 
2. The default calibre show up with few samples books (without the need to login) 
3. You could alternative need to login to access other libraries. 

> --enable-auth, --disable-auth
> 
>     Password based authentication to access the server. Normally, the server is unrestricted, allowing anyone to access it. You can restrict access to predefined users with this option. By default, this option is disabled.
> 

https://github.com/iiab/iiab/blob/d599bc8ecc576739d39d06d8a59c4395ecaaa714/roles/calibre/templates/calibre-serve.service.j2#L9

holta, 2018-08-09:
@arky do you believe there's a way for Calibre to display books without login (at http://box:8080 for now) ...at the same time that teachers add/remove/convert books and fix their medadata using a username and password?

arky, 2018-08-09:
@holta I am familiar with Calibre-Server. Hope this is possible.

holta, 2018-08-09:
> @holta I am not very familiar with Calibre-Server. Hope this is possible.

Agreed.  Let's make this a high priority for [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) &mdash; whether calibre-web (#816) is necessary or not.

arky, 2018-08-14:
@holta  This particular feature is available in calibre-web. It allows public shelf to be accessible to anonymous users.  

https://github.com/arky/iiab/tree/calibre-web

holta, 2018-08-15:
Closing as calibre-web is taking care of this in #816

closed 2018-08-15 by holta