iiab/iiab #994: Should IIAB support NOOBS to broaden participation?

holta, 2018-08-13:
Should we consider supporting Raspbian [NOOBS and its very odd partition tables](https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained) to embrace a substantially broader global demographic, committing more deeply to reduce barriers to entry, welcoming the mass preponderance of Windows users & all?

(Or conversely is NOOBS' partitioning something we just cannot touch, even with a 10-foot pole :-)  e.g. the current glitch arising from "tune2fs -m 1 /dev/mmcblk0p2" on NOOBS (about halfway down IIAB 6.6's installer @ http://download.iiab.io/6.6/install.txt) can easily be fixed using "tune2fs -m 1 /dev/mmcblk0p7" _if NOOBS should even be considered at all?_

floydianslips, 2018-08-20:
This seems like a can of worms that would be extremely difficult to support.

holta, 2018-12-10:
Too expensive to support!

closed 2018-12-10 by holta