iiab/iiab #1002: Update iiab-menu with all recent Content Pack descriptions/logos

holta, 2018-08-16:
https://github.com/iiab/iiab-menu

holta, 2018-08-23:
FYI both critical directories ([images](https://github.com/iiab/iiab-menu/tree/master/menu-files/images) for logos and [menu-defs](https://github.com/iiab/iiab-menu/tree/master/menu-files/menu-defs) for descriptions) are contained within https://github.com/iiab/iiab-menu/tree/master/menu-files

arky, 2018-08-31:
Adding this to UX/UI task list for #1055

holta, 2018-09-03:
See @floydianslips' PR https://github.com/iiab/iiab-menu/pull/7

holta, 2018-12-10:
Tangentially related: #1031

holta, 2019-01-17:
Appears largely done thanks to @tim-moody & @georgejhunt.

And needs to be QA'd in coming months against Haiti's, Mexico's, Peru's (etc!) actual needs.

tim-moody, 2019-01-17:
I want to deprecate all the duplicates.

tim-moody, 2019-02-10:
not sure what remains here.

holta, 2019-02-10:
> not sure what remains here.

For IIAB 6.7, let's just orient folks on how to:
- manually edit [menu def] descriptions & logos
- contact us via http://FAQ.IIAB.IO
- i'll start here overnight, within #1410

For IIAB 7.0 let's start trying to:
- facilitate auto-upstreaming the best curators' [menu def] descriptions & logos
- tabulate & display popularity/quality/usefulness of diff Content Packs & IIAB Apps e.g. for diff ages/curriculums/medical environments
- "reputationally reward" the best curators/reviewers, who share their best _(thereby greatly helping others)_

holta, 2019-02-13:
Better descriptions are needed (and available) for many Content Packs, but as [outlined above](https://github.com/iiab/iiab/issues/1002#issuecomment-462183401) doing this job well takes time, i.e. building relationships with the best curatorial volunteers worldwide &mdash; and so should be a focus of our next release cycle ([IIAB 7.0](https://github.com/iiab/iiab/milestone/5)).

holta, 2019-03-02:
Related: (to field operators & implementers' desire for more curatorial control)

https://github.com/iiab/iiab/issues/828#issuecomment-468933366 Copy Content Packs to/from USB sticks/storage
#857 Auto-display of ZIM files when Teachers insert USB sticks/drives
#1530 Calibre-Web: fix cover/title thumbnails for PDF's and other docs/books
#1538 How does a teacher install a ZIM file from Windows to IIAB? (using a FAT/FAT32 USB stick if poss?)

holta, 2019-07-18:
We still have a long way to go (#1831, possibly embellished by #1866 etc) &mdash; but thankfully the situation for English/Spanish is substantially improving thanks to PR's https://github.com/iiab/iiab-admin-console/pull/242 !

holta, 2020-01-04:
FYI...

1) @tim-moody is working hard on similar/such things, related to stuff like #2089 and #2090.
2) Prior to [IIAB 7.1](https://github.com/iiab/iiab/milestone/6)'s anticipated 2020-01-30 release, I have to submit a small number of Mexico menu descriptions that emerged from the Oaxaca State, Mexico work with ~10 schools a month ago.

holta, 2020-04-21:
FYI @tim-moody [uploaded](https://github.com/iiab-share/js-menu-files/commits/master) several files used in South Mexico schools, to GitHub here:
https://github.com/iiab-share/js-menu-files/tree/master/menu-defs

And a 4th such file (es-kalite.html, fixing KA Lite's top-level menu structure) should hopefully be uploaded soon:
https://github.com/iiab-share/js-menu-files/tree/master/menu-defs

holta, 2020-04-22:
> And a 4th such file (es-kalite.html, fixing KA Lite's top-level navigation/menu structure) should hopefully be uploaded soon!

Done:
https://github.com/iiab-share/js-menu-files/blob/master/menu-defs/es-kalite.html

Likely good enough for IIAB 7.1