iiab/iiab #1006: OpenVPN support using ssh public keys of proven support volunteers

holta, 2018-08-16:
As discussed earlier today on community/team call @ http://minutes.iiab.io ssh public keys for ~4 proven support volunteers should be added to IIAB images when OpenVPN playbook is installed.

Will need to be documented very publicly @ http://wiki.laptop.org/go/IIAB/Security

@georgejhunt may take a crack at implementing this.

georgejhunt, 2018-08-16:
IIAB-FACTORY is missing @jvonau's public key.

ref:https://github.com/iiab/iiab-factory/blob/master/box/keys/developers_authorized_keys

holta, 2018-08-20:
@georgejhunt can you go ahead with 3 keys?  @jvonau's key can be added later as necessary.

holta, 2018-08-20:
Resolved by PRs #1013, #1014

closed 2018-08-20 by holta

holta, 2018-08-20:
Public-facing documentation at http://wiki.iiab.io/Security has been revised accordingly &mdash; please all review that, Thanks!