iiab/iiab #1010: Integration of hospitalrun.io  to IIAB

kananigit, 2018-08-18:
[Hospitalrun](http://hospitalrun.io/)  is an application for remote clinics which help them run operations efficiently and is made i such a way that it can sync its data to an online database when online. Its runs on Node.js, Couch DB database and uses ember cli.

@arky is willing to write a playbook to make the integration possible and asked for the issue to be assigned to him.

holta, 2018-08-20:
@arky can you confirm & explain more when you have time in coming weeks/months?  No rush!

arky, 2018-08-29:
HospitalRun Server is written using Node.Js. It helps clinics manage their inventory, patient listing, schedule tasks such as requests for doctors, lab tests and billing. 

The system is lightweight node.js server, uses a distributed couchdb backend with elasticSearch and logstash for log management. 

There is ember.js frontend for mobile devices, content is stored on mobile devices using pouchdb when offline and synchronized with HospitalRun server using API calls.   

[image]
[image]
[image]
[image]
[image]