iiab/iiab #1028: Implementers R Us: quick wins allowing rapid-fire installation by almost anyone

holta, 2018-08-23:
Implementers want uninterrupted installs (or at the very least, as streamlined and as rapid as possible).

Here are just a few such related issues &mdash; to unlock WYSIWIG rapid provisioning of Internet-in-a-Box's &mdash; across entirely new demographics in 2019:

- [ ] #317 Install OpenVPN/TeamViewer/support tool(s) BEFORE installs begin
- [ ] #872 Reboot + Auto-Continue...IIAB Install Scripts (for OS updates etc)
- [ ] #958 Non-ASCII Characters in install.txt Bash Script
- [ ] #963, https://github.com/iiab/iiab-factory/pull/33 Host install.txt on Github
- [ ] #994 Should IIAB support NOOBS to broaden participation?
- [x] PRs #996, #997, #998, #1000, #1001, #1003, #1004 making the [openvpn playbook/role](https://github.com/iiab/iiab/tree/master/roles/openvpn) more readable, while implementing variable openvpn_handle

jvonau, 2020-09-19:
#1656 iiab/iiab-factory#134

jvonau, 2021-08-17:
Are we(?) there yet?

holta, 2021-08-18:
> Are we(?) there yet?

Absolutely huge progress since 3 year ago in August 2018, for sure!

But "I want what she has" rapid-fire installation of Learning Bouquets is the next steps &mdash; to properly de-bureaucratize.

So the spirit of open learning is not all talk.  But brings "explosive" grassroots action/traction to highly motivated but much-lesser-skilled folk in every country.

jvonau, 2022-04-05:
stale

holta, 2022-04-05:
> stale

And yet ever-present, very real challenges...

jvonau, 2022-09-14:
Yet the single most annoying issue is the forced reboot to cover if RasPiOS's kernel was updated when apt upgrade is ran. Simple solution I mentioned years ago, run apt upgrade last in 'iiab'. Still prime the cache with apt update but should any new package installs require the newest packages of those already installed those will get pulled in anyway. Once iiab/admin are installed what packages that have not been updated will get picked up, nothing really changing other than the kernel might get updated last before the final reboot question. The RasPiOS running kernel/on disk issue is avoided till the very end, but the check for that situation could occur sooner.