iiab/iiab #1031: sampler.html page (used by http://box/home/index.html OOB) needs an overhaul!

holta, 2018-08-23:
This default template for http://box on new installs of IIAB (sampler.html) needs a spring cleaning in coming months, given the many content improvements that have arisen over the past 1.5 years, since @floydianslips worked hard to create this.

Background: line 49 of [./cp-menus](https://github.com/iiab/iiab-menu/blob/master/cp-menus#L49) copies [sampler.html](https://github.com/iiab/iiab-menu/blob/master/samples/sampler.html) into place at /library/www/html/home/index.html during any fresh install of IIAB.

arky, 2018-08-27:
The JavaScript files to detect the browser and rendering menu's needs to be overhauled as well. During one technical demo I found that page took forever to render on  Microsoft surface laptop.   

https://github.com/iiab/iiab-menu/blob/c60dc5ea00c05b568792c817d58bcaa181db9c51/menu-files/js/iiab-menu.js

arky, 2018-08-31:
Adding this to UI/UX refresh task list for #1055

holta, 2018-09-08:
> @holta Please assign this to me, I can quickly fix some bootstrap fixes like responsive layouts

Done.  @darkenvy please assist with any visual/UX work here where you can!

arky, 2018-09-08:
@holta Thanks! 

@darkenvy  Let me know if you have some time to work on it. There are quite a few task here from UI/UX refresh to testing in various devices.

tim-moody, 2018-09-09:
> During one technical demo I found that page took forever to render on Microsoft surface laptop.

What browser? How long is forever? Did it take a similar amount of time for subsequent access? Was the server local or around the world? On a 5 year old windows 10 desktop with chrome, I find a new 15 item menu can take up to 5 seconds to render from a local server on first access, but subsequent access is sub-second because icons and json are then cached.

tim-moody, 2018-09-18:
@arky where is the bootstrap 4 code?

tim-moody, 2018-09-29:
@arky missed your answer.

holta, 2018-12-10:
Tangentially related: #1002, PR https://github.com/iiab/iiab-admin-console/pull/95

holta, 2019-01-17:
@tim-moody are you able to remove sampler.html entirely early next week?  (As we decided to run with during today's call, http://minutes.iiab.io).

Would be really great if you have time!

CALL RECAP for all...: Going forward newly-installed IIAB's will just show (at http://box) those content packs & apps that the implementer has chosen during installation &mdash; e.g. in most cases that'll be {MIN-sized, MEDIUM-sized or BIG-sized} selections as defined by http://wiki.laptop.org/go/IIAB/local_vars.yml

( Of course more advanced installers can [and do] customize their /etc/iiab/local_vars.yml to include different varieties of OLPC apps/services &mdash; see the overall listing of ~20 available IIAB apps/services within http://FAQ.IIAB.IO here: http://wiki.laptop.org/go/IIAB/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F )

holta, 2019-01-17:
> @tim-moody are you able to remove sampler.html entirely early next week? (As we decided to run with during today's call, http://minutes.iiab.io).

( Or @georgejhunt if you have time to do this, either way! )

holta, 2019-01-17:
> I think we should add a dummy module analoglous to test.zim so we have a live link as a module sampler. en-oob.json + test zim + test module + all installed services + credits = new sampler

Consider en-blockly-games from http://oer2go.org/viewmod/en-blockly-games ?

As it's only 4.4MB (smaller than all others, and genuinely used by some communities out there!)

holta, 2019-02-06:
No longer relevant, as new Dynamic Menuing (drag & drop) largely obviates the need for sampler.html

closed 2019-02-06 by holta

holta, 2019-02-06:
Let's recruit more strong mentors to make this happen properly if we want to be a full part of this in 2020~

Internet-in-a-Box will be taken much more seriously if we do!