iiab/iiab #1034: Can Vector OSM search more than 1000 cities?

holta, 2018-08-25:
See cities1000.sqlite (25 MB) as documented near the top of https://github.com/iiab/iiab-factory/blob/master/content/vector-tiles/README.md

Refs: #877, #1033, PR iiab/iiab-factory#39

holta, 2018-08-25:
Question: is there hope for 10,000-or-so searchable cities...and/or countries...and/or oceans/seas/lakes/rivers/mountains/deserts?

georgejhunt, 2018-08-25:
see Experiments #13 in
https://github.com/iiab/iiab-factory/blob/master/content/vector-tiles/Design-Decisions.md

holta, 2018-08-25:
@georgejhunt can you clarify exactly (or roughly) how many cities are in searchable cities1000.sqlite within the zipfiles at http://download.iiab.io/content/OSM/vector-tiles/ ?