iiab/iiab #1035: Can OSM Vector Maps fill the entire screen?

holta, 2018-08-25:
@georgejhunt let's try to examine if Vector-based Map Packs can be made to fill the entire browser screen?

Currently half the screen is wasted (it's all white) on many computers and smartphones e.g. in examples like these two:

http://medbox.iiab.me/modules/en-osm-omt-min/
http://medbox.iiab.me/modules/en-osm-omt-central-am/

Refs: #877 #1033 #1034

holta, 2018-08-26:
Moving to IIAB 6.6 just in case a miracle cure is possible in coming days before final release.

(If this is impossible, we'll try again early in the IIAB 6.7 release cycle.)

georgejhunt, 2018-08-27:
I wonder what version you were testing.  I downloaded the d.iiab.io min,
and it appeared to occupy the whole width of the browser. I think it is up
the the user to maximize the browser.

holta, 2018-08-27:
> I wonder what version you were testing.  I downloaded the d.iiab.io min, and it appeared to occupy the whole width of the browser.

Confirmed, this is great news!  To make testing easier, I republished the 2018-07-07 versions of [en-osm-omt-min.zip](http://download.iiab.io/content/OSM/vector-tiles/en-osm-omt-min.zip) and [en-osm-omt-central-am.zip](http://download.iiab.io/content/OSM/vector-tiles/en-osm-omt-central-am.zip) to be viewable at [http://] iiab.me/modules/en-osm-omt-min/ and iiab.me/modules/en-osm-omt-central-am/ (I do not have access to medbox.iiab.me so let's ask @tim-moody to likewise update its 2 stale OSM demos later in September when he returns!)

_Five OSM viewing bugs in order of priority, when we can &mdash; all 5 occur when viewing page the latest 2018-07-07 version of http://box/modules/en-osm-omt-min/ :_

1. Bottom 15% of browser windows remains white on Chrome and Firefox, on both Windows and Linux client machines, with large screens.  _(Can we make the colored map truly fill the browser window, including the full vertical span of the browser window?)_
2. The top 10% of the browser window remains white (in addition to the bottom 15%) on Chrome and Firefox on an Android 7 phone, when you first open the page (subsequently when you zoom in, this top 10% of the browser window starts being used properly, to display actual/colored maps).
3. The initial map view (when you first arrive at the page) should display Zoom Level 2 (or possible even Zoom Level 3??) so the large white-ish areas (on the left and right of the colored map, on any LCD screen or laptop) cease to exist.
4. Searching for any city on an Android 7 smartphone places the city _"just out-of-view" off the bottom-right of the phone's screen._  Mysteriously: this problem does NOT happen on desktop version of the very same browsers (Chrome and Firefox) when the browser window has been narrowed to show only 200 pixels or so!
5. On an Android 7 phone that is normally very fast, Chrome and Firefox run very slowly &mdash; freezing and crashing both these browsers quite regularly &mdash; when zooming into  page.

holta, 2018-08-29:
Ref: https://github.com/iiab/iiab-factory/pull/42 "remember how to zipup", which creates...

https://github.com/iiab/iiab-factory/blob/master/content/vector-tiles/zipup

holta, 2018-08-30:
Today's community/team call (http://minutes.iiab.io) showcased @georgejhunt massive progress here in recent days:

- NEW en-osm-omt-min (1.5GB .zip) contains everything you need, now published in: http://download.iiab.io/content/OSM/vector-tiles/
- This revised code will be published in coming hours to: https://github.com/iiab/maps
- @tim-moody hopes to update both these live URL's soon in coming days:
  - http://medbox.iiab.me/modules/en-osm-omt-min/ (to version 2018-08-30)
  - http://medbox.iiab.me/modules/en-osm-omt-central-am/ (to version 2018-07-07)

holta, 2018-08-30:
@georgejhunt can I close this as good enough for IIAB 6.6 now?

(And/or is further work ongoing here for IIAB 6.7?)

tim-moody, 2018-08-30:
I wouldn't close until it is in the repo.

holta, 2018-09-08:
@georgejhunt as IIAB 6.6 is not being released in coming hours, can you publish your work to https://github.com/iiab/maps as @tim-moody requested?

@holta wrote:
> @georgejhunt can I close this as good enough for IIAB 6.6 now?
>
> (And/or is further work ongoing here for IIAB 6.7?)

@tim-moody wrote:
> I wouldn't close until it is in the repo.