iiab/iiab #1036: Keep OSM from zooming past Level 10, when no data?

holta, 2018-08-26:
@floydianslips asks @georgejhunt about the new vector-based OSM:

> Is there a way to stop the map from zooming in past "5km" on areas which are not of higher zoom. The map just goes to a whiteout when zooming in closer than "5km" ?

Refs: #877 #1033 #1034 #1035

holta, 2018-08-30:
@georgejhunt is looking into this.

I had thought this was nearly impossible (or certainly an overwhelmingly difficult problem, to get the bounding box correct, no matter which regional dataset(s) are added, as well as working cleanly with most all browsers).

But George & @tim-moody reviewed a few options during this morning's community/team call (http://minutes.iiab.io) such that this might actually be solvable in coming months?

holta, 2018-10-08:
Oops @georgejhunt we forgot to ask about this during https://2018.stateofthemap.us is Detroit!

(If inspiration strikes in coming months, let's shout on the OSM mailing lists to see how others solve this...)

holta, 2019-05-04:
This question has become central to Satellite vs. OSM views, as part of IIAB 7.0 and/or 7.1, i.e. delivering a user interface for http://box/maps that teachers around the world can quick learn & explain, even with the most minimal (or non-existent) training!

holta, 2019-05-10:
A central UX design question — almost as important as the need for a large/colorful button to toggle between "Satellite" view and "Map" View — but it would be extremely challenging to make zooming/panning restrictions elegant, especially before [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) ships in coming weeks!

Those tasks that are too difficult should be deferred to the IIAB 7.1 release cycle, just like this one that will also be in extremely high demand, if it's achievable:

#1644 microSD card pricing has crashed, so OSM map packs could contain continent-wide satellite data, target IIAB 7.1

holta, 2019-05-16:
1) @georgejhunt has made a huge amount of progress here over the past week!

   Unclear whether this will land in IIAB 7.0 or IIAB 7.1 but he'll help us figure that out in coming weeks as we seek to reduce the "change rate" of [IIAB 7.0](https://github.com/iiab/iiab/wiki/IIAB-7.0-Release-Notes) pre-releases now (towards a full feature freeze very soon!)

2) George has also made incredible progress arranging an early prototype that permits IIAB implementers to pre-download satellite photo regions of ~15 km across &mdash; this would allow implementers to pre-install satellite photo coverage around every school/clinic/library/orphanage/etc they want.

   Very possibly this will be ready & good enough for IIAB 7.0 or certainly 7.1...as Sentinel's "anonymous ftp" site does not allow you to download more than ten 256x256 tiles per second.  In other words, it might possibly be too time-consuming to download satellite imagery up to Level 13 across entire continents, as outline below?  Though perhaps "smaller continents" like Central America might possibly be doable &mdash; time will tell in coming weeks/months what's realistic...

Related:

#1644 "microSD card pricing has crashed, so OSM map packs could contain continent-wide satellite data, target IIAB 7.1"

holta, 2019-05-17:
Recap from this morning: @georgejhunt explained during today's IIAB community/team call (http://minutes.iiab.io) that he now has experimental back-end code towards blocking users from "overzooming".

He clarified that this is a higher priority than restraining users from "panning off the edge of the world".

Reason: users who pan/slide/drag their map laterally (e.g. north, south, east or west) tend to quickly understand (that there's no more detailed map/satellite data in that region...) and so tend to recover on their own (usually!)

holta, 2019-05-23:
FYI overzoom blocking is contained within George’s latest prototype Map Pack(s) he’s uploading to archive.org

Please give us a couple more weeks until this goes into public test as part of IIAB 7.0 pre-released and starts to resemble:
- http://iiab.net/osm-vector as spec'd by http://iiab.net/bboxes
- http://iiab.net/map (evolving)

georgejhunt, 2019-05-24:
Thanks for reminding me on the call about my demo machine at iiab.net. I
installed v0.2 versions of san_jose and central_america on that droplet.
They are accessible from the home menu.

As an aside, the wget of the packages from archive.org happened at a rate
of 66mB per second. So I don't think we need to worry about the upstream
bandwidth at archive.org. That would supply 10 clients at the rate I can
download stuff.

holta, 2019-05-24:
> demo machine at iiab.net. I installed v0.2 versions of san_jose and central_america

http://iiab.net offers the link http://iiab.net//osm-vector-maps/en-osm-omt_central_america_2017-07-03_v0.2/ which shows very sharp satellite photos around Venezuela/Brazil & Wyoming &mdash; but very blurry satellite photos in most other areas.

This is before zooming in at all.

Do you know why?  I confirmed this issue is repeatable across different browsers and different client machines.

georgejhunt, 2019-05-24:
Certainly I do not know the answer. But when I was creating the "ocean"
raster tiles for version 1, I remember that the process involved dropping
pixels for the lower resolutions. I'm not concerned as much about fuzzy
images zoomed all the way out.  I'm more concerned about the color scheme,
and the continuity to higher resolution, where the detail can be useful

holta, 2019-06-18:
@georgejhunt I know there's been a ton of progress in recent weeks, but can you just clarify if most (or all!?) of the above issues are resolved?

Thanks !

holta, 2019-12-21:
Good enough for now!

closed 2019-12-21 by holta