iiab/iiab #1037: Add Cockpit Server Manager

arky, 2018-08-26:
Let's use cockpit to manage system services (systemd), network and packages. It uses very little system resources like memory, cpu offloading much of the work to client web browser. The cockpit service socket listens on port 9090.  Cockpit has been introduced in/since RHEL 7, it has been back ported older debian based distros (See https://github.com/iiab/iiab/issues/987 ) 

"Cockpit is a server manager that makes it easy to administer your GNU/Linux servers via a web browser."

https://cockpit-project.org/

holta, 2018-08-30:
@jvonau asks if you can restrict backports (in #987) to only those things required for cockpit?

arky, 2018-08-30:
@jvonau Yeah, Backport packages are apt pinned to prevent problems.  See  https://github.com/iiab/iiab/issues/987#issuecomment-411662832

jvonau, 2020-04-29:
stale

closed 2020-04-29 by holta