iiab/iiab #1040: Calibre-Web: How can teachers save the save language(s) of an e-book?

holta, 2018-08-28:
### Expected behavior and actual behavior.

There must be a way for teachers to edit and/or save an e-book's language using Calibre-Web &mdash; but the UX is highly non-intuitive if so?

### Steps to reproduce the problem.

- log into http://box/books using Admin/changeme
- click "UPLOAD" on the top of the page; add a PDF book or similar
- enter "English" or "EN" etc into the "Language" textfield
- click Submit button
- use Calibre-Web to browse to the newly uploaded book
- click "Edit metadata"
- sadly the "Language" field is completely blank...

### Specifications like the version of the IIAB, operating system version, or hardware details.

IIAB 6.6/master on Raspbian.

Refs: #816, PR #999, PR #1038, PR #1039

arky, 2018-08-29:
Response from calibre-web developer upstream: 

> There is a known limitiation (to me known). If you want to set the language you have to have at least one book in your calibre library with that language to make this work. So please open calibre with the database you want to use and add a language you want to use to one of the exisiting books. Afterwards you should be able to add further books in Calibre-Web with this language.

holta, 2018-09-03:
Another [great update](https://github.com/janeczku/calibre-web/issues/593#issuecomment-418187996) today from @OzzieIsaacs helping us understand the magnitude & details of this central challenge, deeply affecting all teachers and hence students.

In short, this is not solvable for Internet-in-a-Box 6.6 this week &mdash; so let's defer to IIIAB 6.7

holta, 2018-09-09:
There appears to be INCREDIBLE progress from @OzzieIsaacs @ https://github.com/janeczku/calibre-web/issues/593#issuecomment-419727467

We urgently need to help him test this in coming days!

holta, 2018-09-09:
1) Language selection is confirmed to work!  Certainly when the entire Calibre-Web UX is in English, presumably as a result of $LANG or file /etc/default/locale or output of /usr/bin/locale ?  In any case, @arky let's recruit volunteers to test in Spanish, French and other languages in coming days!

2) Calibre-Web's "Default" bookshelf (http://box/books/shelf/1) fails to appear, with error:

>Internal Server Error
>
>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

CLARIF:
- the server is NOT overloaded
- this error occurs regardless if a book can been tagged as being on the "Default" bookshelf (or not!)
- rebooting does not help

There was no problem however with a manually-created bookshelf like http://192.168.0.121/books/shelf/2 &mdash; FYI I'm testing on Ubuntu 18.04 Desktop machine &mdash; but if it helps I can:
- test other OS's like Debian 9.5 and Raspbian on Raspberry Pi?
- verify log files [which?] if any such would be helpful to @OzzieIsaacs?
- create a new ticket within https://github.com/janeczku/calibre-web/issues/ ?

OzzieIsaacs, 2018-09-10:
- the loginscreen is always in English, the UI language can be set on a per user base (second button from the right). While creating users the language can be set to whatever you want.
-The text for the internal server error is the standard one, up to now it never was overload, it‘s always programming errors. 
- there should be no ‚default‘ bookshelf
- Theresia a logfile in the root folder of the application named Calibre-web.log.  Please file a issue in the Calibre-web repro and I‘ll check.

holta, 2018-09-10:
> the loginscreen is always in English, the UI language can be set on a per user base (second button from the right). While creating users the language can be set to whatever you want.

Confirmed!

> The text for the internal server error is the standard one, up to now it never was overload, it‘s always programming errors.

Good to know.

> there should be no ['Default‘] bookshelf

What might be creating this http://box/books/shelf/1 do you know @arky or @OzzieIsaacs ?

> [There is] a a logfile in the root folder of the application named [calibre-web.log]. Please file [an] issue in the Calibre-web [repo] and I‘ll check.

Done!  Those who can, please see: https://github.com/janeczku/calibre-web/issues/620 "500 Internal Server Error when you click on "Default" bookshelf (on main page)"

holta, 2018-09-11:
@arky both Calibre-Web issues appear under control, profound thanks to @OzzieIsaacs :

1. So I'm closing this (#1040) for now.
2. Please help me reach out cultivating further testers!

closed 2018-09-11 by holta