iiab/iiab #1041: Allow users to control Calibre-web provisioning

arky, 2018-08-28:
@holta  Please kindly allow users to do provisioning with 'calibreweb_provision' variable. This is pretty standard practice in configuration management deployments. 

Just checking if metadata.db /  app.db exits seems like a good idea but I would recommend using variables to allow more fine grained control of deployments.  

https://github.com/iiab/iiab/commit/2e6ab33874774205f16a0de81eef2180eb25eb1e

holta, 2018-08-28:
As written the default `calibreweb_provision` was too dangerous, in that it would have caused deletion of e-books metadata often among implementers who often run "iiab-install --reinstall" as they're adding content, racing like crazy to deliver Internet-in-a-Box units to schools.

This is the reason that the majority of our IIAB services/apps use a trigger file, that allows sysadmins to force an upgrade of that IIAB service/app &mdash; without harming others &mdash; even if they have zero understanding of Ansible: http://wiki.laptop.org/go/IIAB/FAQ#Can_I_upgrade_or_reinstall_server_apps.3F

But if an additional `calibreweb_provision` flag is truly also needed, in addition to a trigger file (in this case `/library/calibre-web/metadata.db`) then let's just make sure that is designed carefully to be failsafe, so both use cases don't collide with each other~

holta, 2018-08-28:
Here are the 2 most common upgrade/clean install scenarios requested by IIAB implementers:

1) UPGRADE by OVERLAYING (some IIAB service/app) to the latest version WITHOUT losing content or settings/accounts/passwords!

2) UPGRADE by WIPING (some IIAB service/app) to the latest version WHILE forcing all settings/accounts/passwords to the default &mdash; _even if that means losing existing content!_  HUGE BONUS: if the implementer is _forced_ to create a complete backup prior to proceeding, e.g. by moving the entire directory of content + settings (if such exists) then informed consent is 100% confirmed.

Separately here are a few other scenarios that arise on rare occasion, but are not nearly as common:

3) REINSTALL by OVERLAYING the existing/old version (of some IIAB service/app) WITHOUT losing content or settings/accounts/passwords!

4) REINSTALL by WIPING the existing/old version (of some IIAB service/app) WHILE forcing all settings/accounts/passwords to the default &mdash; _even if that means losing existing content!_

Both above are more rare, but can occur when an older software version proves more reliable/more loved than a recent version.  Finally some folks dream of logical/mathematical validation of existing installations, which is not easy, but might be a valuable goal in coming years as security concerns grow:

5) AUDIT the installation of (some IIAB service/app) without upgrading it, without touching content nor its settings/accounts/passwords &mdash; i.e. don't touch anything but sanity check it's working.

Understanding patterns like the ~5 above (which are truly needed! which rarely occur!) will help us look to the future (re)designing http://FAQ.IIAB.IO#Can_I_upgrade_or_reinstall_server_apps.3F in service of teachers/implementers pragmatic needs &mdash; serving the public across all IIAB services/apps.

_EXAMPLE: we used to think offline upgrades and offline reinstall of IIAB apps would be centrally important.  We were 100% wrong in the end &mdash; waiting for Ansible (and similar configuration exercises) are not tolerated in the field, when teachers / medical clinicians are overwhelmed with their jobs already.  So we've recalibrated our assumptions over the past 6 years &mdash; allowing our engineering efforts to help many more people when applied elsewhere.  (Just one example of the need to keep grounding our system designs & UX designs on proven usage patterns~)_

arky, 2018-08-29:
Great writeup, this is basically what I implemented using 'calibreweb_provision' variables. These best practices should be followed by all the roles in IIAB.  

1. UPGRADE by OVERLAYING:  when 'calibreweb_provision = False' the role basically retains all the existing content and settings while updating the calibre-web to latest master. 

2. UPGRADE by WIPING:  when 'calibreweb_provision = True' the role backups existing databases with 'backup=yes' and then copies the provisioned version of app.db, metadata.db. 

The Ansible backup option backups the database with timestamp. Using the combination of 'calbreweb_provision' and 'calibreweb_*database*' variables, we can address lot of problems that you mentioned and also automate large scale deployments I do.

holta, 2018-09-26:
SEE near-duplicate tkt:

#799 "Standardize upgrade/reinstall of IIAB services (server apps)"

jvonau, 2020-09-19:
This layout is more or less in place with the iiab_state.yml file and would be complemented by #2503 with the use of provisioned in the iiab_state.yml file. I'd have to double back, I was just looking at mysql related apps for 2503 and see what calibreweb does during install.yml but fits the model invisioned with the split install from provisioning.