iiab/iiab #1055: 3-5 hours tasks for Google Code-in (application begins 2018-09-06; students begin 2018-10-23)

holta, 2018-08-30:
Let's add to our current [IIAB 6.7 Milestone](https://github.com/iiab/iiab/milestone/4) with clearly-articulated / bite-sized [Google Code-in](https://codein.withgoogle.com) ideas filling out @arky's outline here:

Technical
---------
* Ansible Enhancements: Doing code review to older less used roles for
instance could be very help.

* Ansible CI/Testing: Writing Python units tests and fixing the Ansible
2.6 CI broken TravisCI integration could be done in few hours.

* Testing: Various network configuration and on distributions such as
Debian, CentOS etc.

Design
------
* UI/UX: Implementing / Rebooting home pages using bootstrap 4 would be
great.

* Design / Marketing / Websites: We need a great logo, color scheme for
the project,  a good copy for the website wont' hurt either.

Documentation
--------------
* Documentation: Cleaning up and writing new wiki pages for new users
and getting started guide for new contributors is good start. That's how
I started contributing.

arky, 2018-08-30:
@holta  Can you we try out Github Projects to build this out?

https://help.github.com/articles/about-project-boards/

holta, 2018-08-30:
> https://help.github.com/articles/about-project-boards/

GitHub Project Boards might not be the world's greatest collaboratory (compared to [Trello](https://trello.com/), [Slack](https://slack.com/), [Asana](https://asana.com/), [Discourse](https://www.discourse.org/) etc) but we definitely need something here, so yes let's please investigate!

arky, 2018-08-31:
@holta  Great, Can you please set up a new project board for 'Google Code-in tasks' and add permissions for the organization to add notes to it. Thanks 

https://help.github.com/articles/project-board-permissions-for-an-organization/

arky, 2018-08-31:
@holta I went through all the open bugs and reference all bugs related to Design section of my outline. Perhaps we can add these bugs to the project page when you create one. Thanks!

tim-moody, 2018-08-31:
Surely there's a better home for this sort of project planning than inside an issue, which will go away at some point.

tim-moody, 2018-09-01:
I agree it looks promising. Why don't you take it on.

arky, 2018-09-01:
@tim-moody Sure, At the moment I don't have permission to create/edit project boards.

holta, 2018-09-04:
@arky:

1) I'd misunderstood that we (in fact) must be part of GSoC (Google Summer of Code) starting in January 2019 before we qualify for GCi (Google Code-in) later in 2019.  Let's not give up however, as there's important prep we can do in the 4 months between now and January~

2) Does something like GitHub Project Board https://github.com/orgs/iiab/projects/1 get us off on the right track here, preparing for January, pulling together concrete student projects from #1055 here and https://github.com/iiab/iiab/milestone/4 etc?  I set you (@arky) as an administrator for the Project Board, and made the entire thing public, but LMK if "Basic Kanban" is not enough, among the choices at https://help.github.com/articles/about-project-boards/#templates-for-project-boards

tim-moody, 2018-09-04:
BSF uses http://shareit-taiga.cleverapps.io, to mention an alternative.

holta, 2018-10-15:
Legal/licensing work such as https://github.com/iiab/iiab/issues/1229 should be on the list too &mdash; as our scraping community grows up &mdash; to have as broad a _public interest impact_ as possible.

holta, 2018-10-16:
Another self-contained-yet-critically-important task for potential new contributors:

#1231 [Testing/CI] Testing framework for Ansible roles

arky, 2018-10-16:
@holta Can you please create a project board. I don't have permission to create/edit project boards.

holta, 2018-10-16:
> @holta Can you please create a project board. I don't have permission to create/edit project boards.

@arky this is the one I created about a month ago giving you admin access:

https://github.com/orgs/iiab/projects/1

Is that sufficient?

arky, 2018-10-16:
I think it is better to create a project board on the repo itself. Can you please create one here: 
https://github.com/iiab/iiab/projects/

tim-moody, 2018-10-16:
I would create a board for each project or at least call this Google Code-in Projects.

tim-moody, 2018-10-16:
Sorry I missed the name GSoC. the url seems just to have project number rather than name.
I like task hierarchy so experimented with headings at https://github.com/tim-moody/iiab-menu/projects/1.
I find it hard to get the right granularity,

holta, 2018-10-16:
> I think it is better to create a project board on the repo itself. Can you please create one here:
> https://github.com/iiab/iiab/projects/

Does https://github.com/iiab/iiab/projects/1 work?

I can make further modifs later as nec.

holta, 2018-11-14:
We have 2 months to prepare for GSoC's 15th anniversary program, just announced, with Org applications opening January 15th, 2019 !

https://opensource.googleblog.com/2018/11/google-summer-of-code-15-years-strong.html
https://developers.google.com/open-source/gsoc/timeline
https://developers.google.com/open-source/gsoc/resources/marketing
https://developers.google.com/open-source/gsoc/faq
https://developers.google.com/open-source/gsoc/videos
https://summerofcode.withgoogle.com/archive/ [archive of 2018 GSoC]

holta, 2019-01-16:
@aidan-fitz let's talk live in the coming week about this, if you have time!

aidan-fitz, 2019-01-16:
My schedule this next week is uncertain, but I'm available on weekends.

holta, 2019-02-06:
Let's recruit more strong mentors to make this happen properly if we want to be a full part of this in 2020~

Internet-in-a-Box will be taken much more seriously if we do!

holta, 2019-03-14:
Let's also keep an eye on the band new "Google Season of Docs" program:

https://opensource.googleblog.com/2019/03/introducing-season-of-docs.html
https://developers.google.com/season-of-docs/

> "Let's bring open source and technical writer communities together, to the benefit of both. Together we raise awareness of open source, of docs, and of technical writing."

Excerpts from [Timeline](https://developers.google.com/season-of-docs/docs/timeline):

- Apr 2-23, 2019 ~ Mentoring organizations submit applications to Google
- Apr 30, 2019 ~ Organizations announced
- Apr 30 - May 28, 2019 ~ Interested technical writers discuss project ideas with mentoring organizations
- May 29 - Jun 28, 2019 ~ Technical writers submit applications
- Jun 30, 2019 ~ Technical writing projects announced
- Jun 28 - Jul 23, 2019 ~ Proposal review period for organizations
- Jul 23 - Jul 30, 2019 ~ Proposal review period for Google
- Jul 30, 2019 ~ Google announces the accepted technical writer projects
- Aug 1 - Sep 1, 2019 ~ Community bonding
- Sep 2 - Nov 22, 2019 ~ Doc development
- Nov 25 - 29, 2019 ~ Project finalization and evaluation
- Dec 2 - 6, 2019 ~ Mentors submit evaluations
- Dec 10, 2019 ~ Results announced
- Feb 24 - 28, 2020 ~ Final week for long-running projects
- March 6, 2020 ~ Google adds long-running projects to the results of "Season of Docs 2019"

aidan-fitz, 2019-03-14:
> Let's recruit more strong mentors to make this happen properly if we want to be a full part of this in 2020~
> 
> Internet-in-a-Box will be taken much more seriously if we do!

GitHub's [open source guide] has a page on building community. We should think about following some of the steps on this page if we want to attract more contributors!

[open source guide]: https://opensource.guide/building-community/

jvonau, 2021-07-22:
is this still relevant today?

holta, 2021-07-22:
> is this still relevant today?

Relevant.  But needs a complete rethink for sure.