iiab/iiab #1063: [Marketing] Update internet-in-a-box.org content

arky, 2018-08-31:
Update the content of the main project website. A better description of the project, images of the IIAB hardware, deployments in schools and feature list. 

Task list for https://github.com/iiab/iiab/issues/1055

holta, 2019-03-14:
This cleanup from mid-February is a start, at least helping get our grassroots/global narrative framework in place, so that a modern site with stronger videos/photos can be developed in due course:
- http://iiab.io
- http://internet-in-a-box.org

holta, 2019-10-01:
Tweaked for IIAB 7.0's release to be announced incoming hours, to show Peru's IIAB video.

Of course we can always do better!  And will in future.  So I'm moving this tkt to the IIAB 7.1 Milestone.