iiab/iiab #1066: [CI] TravisCI VM broken since Ansible 2.6.x

arky, 2018-09-01:
TravisCI default script has broken since the Ansible 2.6. Perhaps there is some issues with the default VM of Travis. 

Need to investigate and report upstream to Ansible. 

```
$ ansible --version
ansible 2.6.3
  config file = /home/travis/build/arky/iiab/ansible.cfg
  configured module search path = [u'/home/travis/.ansible/plugins/modules', u'/usr/share/ansible/plugins/modules']
  ansible python module location = /home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/ansible
  executable location = /home/travis/virtualenv/python2.7.14/bin/ansible
  python version = 2.7.14 (default, Nov 28 2017, 18:46:37) [GCC 4.8.4]
Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.14/bin/ansible", line 161, in <module>
    shutil.rmtree(C.DEFAULT_LOCAL_TMP, True)
AttributeError: 'module' object has no attribute 'DEFAULT_LOCAL_TMP'
The command "ansible --version" failed and exited with 1 during .

```

Task list for #1055

holta, 2018-09-01:
If possible check Ansible 2.7 Beta 1 (just released) too?

( FWIW we will transition IIAB/master from 2.6.x to Ansible 2.7.0 rather quickly, like on/around 2018-10-04: https://docs.ansible.com/ansible/devel/roadmap/ROADMAP_2_7.html )

holta, 2018-10-11:
Further explanation at #1212

arky, 2018-10-14:
Closing this bug, patched in https://github.com/iiab/iiab/pull/1217

arky, 2018-10-14:
Re-opening this bug based on the real source of this problem. 

https://github.com/iiab/iiab/blob/9269a15e84c495c95ea05ce6633c5cc34ec9f35c/ansible.cfg#L6-L7

closed 2018-10-17 by holta