iiab/iiab #1070: Allow remote IIAB Deployments from local computer

arky, 2018-09-01:
Allow IIAB deployments from remote machines (The Ansible Way).  IIAB is installed by running ansible commands on the deployments machine locally. The ideal way is to use ansible from local computer/laptop and ansible runs the IIAB deployments on the target machine.

Why you need this?
 * This is how ansible is designed to work. 
 * Helps you run deployments from your local computer over a network on headless systems like raspberry pi.
 * You can quickly modify/update large number of deployments which physically logging into the machine using SSH
* You can do CI/Unit Testing by creating vagrant virtual machines. 

Task List for #1055

arky, 2018-09-02:
@jvonau Please do share the complete details about this 'os_ver' problem. Let's get this patched upstream quickly.

jvonau, 2020-04-29:
I have been able to boot a rpi and run ansible against the rpi without having ansible installed on the target.

jvonau, 2022-04-05:
close no interest.

holta, 2022-04-05:
> close no interest.

Ok.

(This can be re-opened at any time if someone is motivated to work on this to fit worthy community needs!)

closed 2022-04-05 by holta