iiab/iiab #1078: Diverse/futurist BitTorrent suggestions prior to IIAB 6.6 final release?

holta, 2018-09-02:
Building off #884 -> PR #1011 -> PR #1044 -> PR #1061

holta, 2018-09-21:
RELATED: #977 (Kolibri 0.10.2+ Release Polish...as it evolved to 0.11 and beyond!)

arky, 2018-09-22:
It seems to me that there is no consensus on when/if IPFS  will be adopted to address this issue in upstream projects. 

Perhaps we could only focus on streamlining existing transmission code in this bug until at certain point in future when an alternative solution becomes available.

holta, 2019-05-04:
There is progress here — a basic version of Internet Archive's Decentralized Web services should integrated as part of IIAB 7.0 in coming weeks — and hopefully much more from.@m-anish as part of IIAB 7.1:

Refs: #1546 #1576 PR #1617

mitra42, 2019-10-04:
But note ... we disable IPFS in the Offline Archive code because of the CPU/bandwidth overhead. It also won't work in offline mode, and (at least for our content) is still single-point-of-failure, i.e. has to be able to reach the archive.org servers. They've got a lot on their plate at IPFS, and making it work in low CPU / low bandwidth / single-point-of failure is AFAIK not on anyone's agenda.

BitTorrent / WebTorrent has some potential for these environments, especially for rebroadcast of video between end-user devices, I hope to get a tracker build into the offline archive at some point, but don't have the bandwidth to do so at the moment.