iiab/iiab #1094: Kolibri did not start within 2MIN timeout during BIG install on RPi 3

holta, 2018-09-06:
Should we raise this 2MIN timeout?

Excerpt from /opt/iiab/iiab/iiab-install.log:

> 2018-09-06 09:17:30,026 p=4762 u=root |  TASK [kolibri : Enable & (Re)Start kolibri service] ****************************
2018-09-06 09:19:02,440 p=4762 u=root |  fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kolibri: Job for kolibri.service failed because a timeout was exceeded.\nSee \"systemctl status kolibri.service\" and \"journalctl -xe\" for details.\n"}

If so should we do this in https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri.service.j2 ?

Or in [Lines 42-48 of roles/kolibri/tasks/main.yml](https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/main.yml#L42-L48), with Ansible "async" setting perhaps?

holta, 2018-09-07:
Clarif: this tkt like most/all others will be moved to the [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) release cycle in coming hours.

arky, 2018-09-21:
@holta  This might be caused by kolibri migrations. We need to figure out what migrations are taking most time. Please also assign this task to me. Thanks!

holta, 2019-03-05:
Symptoms are very similar...but not quite the same as?

#1489 Kolibri service is timing out on boot on Ubuntu 18.04 (and possibly other OS's too?)

holta, 2019-03-06:
Let's consider closing this tkt in April 2019, if the issue can't be reproduced.

holta, 2019-05-05:
Re-open as nec!

closed 2019-05-05 by holta

ITGuyClaude, 2019-08-13:
I also had this issue on Ubuntu 18.04, PI 3 (Stretch) and PI4. Changing the Kolibri service to the below resolved it for me.

pi@box:/etc/systemd/system $ cat kolibri.service 
[Unit]
Description=Kolibri

[Service]
Type=oneshot
RemainAfterExit=yes
User=pi
ExecStart=/usr/local/kolibri/bin/kolibri start
ExecStop=/usr/local/kolibri/bin/kolibri stop

[Install]
WantedBy=multi-user.target