iiab/iiab #1107: [FAQ] Default Download Folder on IOS devices

arky, 2018-09-07:
Feedback from field trials of IIAB from Phnom Penh. 

Users with Apple IOS iphone and ipad devices struggled to find the books and videos that they have downloaded from IIAB. AFAIK Apple devices automatically opens the Books (PDFs) in a web browser and doesn't provide an option to save it to local folders but only to iCloud or Dropbox.

I am not familiar with the Apple mobile devices workflows, perhaps someone would write a FAQ for those with IOS device on how to download and access books and videos.

holta, 2018-09-07:
@arky where are they getting most of these PDF's at the moment?

Calibre or Calibre-Web or Nextcloud and/or any other/specific RACHEL-like Content Packs ?

A big problem we've seem is overweight PDF's e.g. those above 5-10MB especially, tending to drive almost everyone crazy &mdash; as they often have no indication how long it will take to arrive in their phone/device &mdash; and whether it will display quickly later or not.

In these cases it's critical to give users/teachers/all advance notice as to how big / how compatible each PDF is *prior* to clicking on it :-)

arky, 2018-09-07:
@holta  Calibre-web using books from Let's Read Asia project. (https://reader.letsreadasia.org) I don't think the size of the documents is not really the issue here. 

I think simple instructions like this were need in my fieldtrial. I wish I knew more about Apple devices.

How to save a PDF on iPhone/iPad. 
https://www.youtube.com/watch?v=a6A1ezWzG4I

arky, 2018-09-08:
(To add more context to this trivial issue. It was super awkward for me to explain that the inability to see where your files are downloaded (or how to download a video) is not a problem with *Internet in a box* per se. It is really important to understand that people who own a smart phone are not always savvy enough to use it properly.)

Based on my limited knowledge, iOS design is very restrictive. It tries to do guess what application to use based on files MIME metadata. I am not aware of any free iOS apps that allow you to view/play local content like books and videos.

So I hope someone in the community could shed the light.