iiab/iiab #1108: Use Handlers for Kolibri Services

arky, 2018-09-08:
Let's refactor Kolibri service to use handlers instead. Please assign this bug to Arky. Thanks! 

https://github.com/iiab/iiab/blob/d1bbffb14f75794689a8b30c925b1f51b2894e76/roles/kolibri/tasks/main.yml#L42-L46

holta, 2018-09-08:
Building on #841 #920 #922 + PRs #912 #916 #918 #1076 / #1077

Consolidated @ checklist #977

jvonau, 2018-09-19:
If the push is to use handlers please re-factor all installed services to use handlers.

arky, 2018-09-21:
Tip from upstream Ansible developers: "Feature Idea: Allow service module to have more states"

https://github.com/ansible/ansible/projects/13#card-5000281

arky, 2019-01-18:
> Tip from upstream Ansible developers: "Feature Idea: Allow service module to have more states"
> 
> https://github.com/ansible/ansible/projects/13#card-5000281

Closing this bug for now. Re-open if needed.

closed 2019-01-18 by arky