iiab/iiab #1109: Clean up RedirectMatch rules

arky, 2018-09-08:
Perhaps we need to investigate IIAB RedirectMatch rules as pointed out by Ivan in https://github.com/iiab/iiab/issues/913#issuecomment-410484137

> Scoping directives to <VirtualHost *:80> has some unholy interaction with the RedirectMatch ^/$ /home in sites-enabled/iiab-homepage.conf (/kolibri opens the python server on port 8009 but /kolibri/ was sending me to /home/) 
https://github.com/iiab/iiab/issues/913#issuecomment-410484137

https://github.com/iiab/iiab/blob/d1bbffb14f75794689a8b30c925b1f51b2894e76/roles/homepage/templates/iiab-homepage.conf#L9

Refer: https://github.com/iiab/iiab/issues/913

holta, 2018-09-12:
Thanks @arky if you're able to research this in coming weeks &mdash; for the benefit of #923 and all playbooks?

holta, 2018-09-26:
Can this build on the RedirectMatch used in PR #1060 "http://box/kiwix was failing" ?

holta, 2019-02-06:
@arky can we close this now that your #913 -> PR #1412 are complete?

Or feel free to re-open as nec!

closed 2019-02-06 by holta