iiab/iiab #1130: Remove Ansible's pink/red warnings. Prep for Ansible 2.8/2.9 deprecations.

holta, 2018-09-13:
Not all of Ansible's pink/red warnings are removable &mdash; indicating too many callouts to bash/sudo etc &mdash; but most will be!

Ref: #1100 (Migrate to Ansible 2.7 on "2018-10-04")

holta, 2018-10-14:
See PR #746 "Speedup by squashing package installs in ansible.cfg (condenses loop)" which is related &mdash; that PR was a real lifesaver (saving 10-20min on each RPi install!) contributed by @jvonau in April 2018.

arky, 2018-10-14:
It seems this is a trivial fix. You shouldn't use 'with_items' looping in package module anymore.

Use 'name:' instead of 'with_items' looping: 

```
package:
    name: "{{ item }}"
    state: present
  with_items:
    - openvpn
    - nmap
```

```
package:
    name: "{{ item }}"
    state: present
  name:
    - openvpn
    - nmap
```

holta, 2018-10-28:
More than 90% of our Ansible 2.7.1 warnings are now gone!

But this FAIL & WARNING now remain: (can others help with these two?)
```
TASK [postgresql : Initialize the postgres db (debuntu)] ***********************
 [WARNING]: Consider using 'become', 'become_method', and 'become_user' rather
than running su

TASK [kolibri : Create kolibri default facility name, admin account and language] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "export KOLIBRI_HOME=\"/library/kolibri\" && \"/usr/local/kolibri/bin/kolibri\" manage provisiondevice --facility \"Kolibri-in-a-Box\" --superusername \"Admin\" --superuserpassword \"changeme\" --preset \"formal\" --language_id \"en\" --verbosity 0 --noinput", "delta": "0:00:01.557827", "end": "2018-10-28 16:21:18.735685", "msg": "non-zero return code", "rc": 1, "start": "2018-10-28 16:21:17.177858", "stderr": "\u001b[37mINFO     Running Kolibri with the following settings: kolibri.deployment.default.settings.base\u001b[0m\n\u001b[33mWARNING  Facility with name Kolibri-in-a-Box already exists, not modifying preset.\u001b[0m\nTraceback (most recent call last):\n  File \"/usr/local/kolibri/bin/kolibri\", line 11, in <module>\n    sys.exit(main())\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 637, in main\n    manage(command, args=django_args)\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 427, in manage\n    execute_from_command_line(argv=argv)\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 364, in execute_from_command_line\n    utility.execute()\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 356, in execute\n    self.fetch_command(subcommand).run_from_argv(self.argv)\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 283, in run_from_argv\n    self.execute(*args, **cmd_options)\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 330, in execute\n    output = self.handle(*args, **options)\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 119, in handle\n    interactive=options['interactive'],\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 72, in create_superuser\n    FacilityUser.objects.create_superuser(username, password)\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/auth/models.py\", line 496, in create_superuser\n    raise ValidationError(_('An account with that username already exists'))\ndjango.core.exceptions.ValidationError: [u'An account with that username already exists']", "stderr_lines": ["\u001b[37mINFO     Running Kolibri with the following settings: kolibri.deployment.default.settings.base\u001b[0m", "\u001b[33mWARNING  Facility with name Kolibri-in-a-Box already exists, not modifying preset.\u001b[0m", "Traceback (most recent call last):", "  File \"/usr/local/kolibri/bin/kolibri\", line 11, in <module>", "    sys.exit(main())", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 637, in main", "    manage(command, args=django_args)", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 427, in manage", "    execute_from_command_line(argv=argv)", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 364, in execute_from_command_line", "    utility.execute()", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 356, in execute", "    self.fetch_command(subcommand).run_from_argv(self.argv)", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 283, in run_from_argv", "    self.execute(*args, **cmd_options)", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 330, in execute", "    output = self.handle(*args, **options)", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 119, in handle", "    interactive=options['interactive'],", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 72, in create_superuser", "    FacilityUser.objects.create_superuser(username, password)", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/auth/models.py\", line 496, in create_superuser", "    raise ValidationError(_('An account with that username already exists'))", "django.core.exceptions.ValidationError: [u'An account with that username already exists']"], "stdout": "", "stdout_lines": []}
...ignoring
```

closed 2018-11-30 by holta