iiab/iiab #1136: kalite install error on fresh 6.6 iiab

ThatMattGuy123, 2018-09-17:
### Expected behavior and actual behavior.
It's supposed to install, it's not.
### Steps to reproduce the problem.
Download current 2018-06-27-raspbian-stretch-lite edition, update/upgrade/dist-upgrade, reboot, update/upgrade/dist-upgrade once more. Install iiab per "curl download.iiab.io/6.6/install.txt | sudo bash" command.
### Specifications like the version of the IIAB, operating system version, or hardware details.
6.6 iiab, 2018-06-27-raspbian-stretch-lite, rpi 3 b+

See screenshot. 

[image]

holta, 2018-09-17:
1) @tim-moody can you explain which ./cp-menus (somewhere between [Lines 55-71](https://github.com/iiab/iiab-menu/blob/master/cp-menus#L55-L71)) appears to be failing in the screenshot above?  With error message:
```
files copies
ERROR at line 1: Unknown command '\Y'.
database created
```

2) @ThatMattGuy123 what size/brand SD are you using?

3) @ThatMattGuy123 your screen looks very different from typical RPi output (quotation marks are missing).  What exact LOCALISATIONS (A. locales, B. default locale, C. time zone, D. Wi-Fi country, E. any other settings) did you set when running raspi-config?

4) @ThatMattGuy123 with an identical OS and updates, I get different (working) output below.  So we need to know more about how your HW/SW/steps-you're-taking which are different from the way others install this:
```
database created
Generating a sharing network.
Generating a sharing network invitation--from me, to me!
Done!
--2018-09-15 00:29:52--  http://pantry.learningequality.org/downloads/ka-lite/0.17/content/contentpacks/en.zip
Resolving pantry.learningequality.org (pantry.learningequality.org)... 162.209.63.41
Connecting to pantry.learningequality.org (pantry.learningequality.org)|162.209.63.41|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 929916955 (887M) [application/zip]
Saving to: ‘en.zip’

en.zip              100%[===================>] 886.84M  8.94MB/s    in 82s

2018-09-15 00:31:15 (10.8 MB/s) - ‘en.zip’ saved [929916955/929916955]

[WARNING] [2018-09-15 00:31:23,151] kalite.i18n.base: No en_icu.js file found in locale_path /library/ka-lite/locale
[WARNING] [2018-09-15 00:31:23,153] kalite.i18n.base: No en_icu.js file found in locale_path /library/ka-lite/locale
[INFO] [2018-09-15 00:31:23,154] kalite.i18n.base: Writing i18nized js file to /library/ka-lite/content/locale/js/i18n/en.js
[INFO] [2018-09-15 00:34:09,664] kalite: Annotating content for language: en, channel: khan
...
```

holta, 2018-09-17:
1) @ThatMattGuy123 can you please paste in the results of these 3 commands?  (To help others try to reproduce the situation exactly, which others have not been able to do so far)
   - more /etc/default/locale
   - ls -l /etc/*time*
   - more /etc/hostapd/hostapd.conf

   ASIDE: Matt was not aware that he needed to start a new shell or window (or reboot) after changing the above LOCALISATIONS.

2) Can you please confirm the errors errors in your screenshot happen on *different* microSD cards?  (Or is it just one make/model of microSD card...or a particular microSD card...that might possibly be dying?)

   Very unfortunately it seems the 64GB microSD card involved in this report has been wiped.  But @ThatMattGuy123 will now try to reproduce these "3" issues on the same microSD card and others...

CONTEXT: Three apparently-very-different issues/errors (on the very same screenshot, as itemized [above](https://github.com/iiab/iiab/issues/1136#issuecomment-422136498)) make this *highly* unusual/mysterious, so we need more background than the usual this time!

ThatMattGuy123, 2018-09-18:
I probably forgot ot reboot aftrer setting the local, country, time zone, etc. After I nuked the rpi it worked worked as expected to install iiab.

holta, 2018-09-18:
I'm unable to reproduce @ThatMattGuy123's 3 issues from yesterday, under essentially identical conditions:

- apostrophes appear on screen as they should
- ./cp-menus completes without error
- both "kalite manage" commands in install.txt complete as they should

So until we know more I attribute yesterday's 3 problems as (likely) caused by an unreliable SD card/filesystem, not directly related to {raspi-config, ./cp-menus, KA Lite}.

closed 2018-09-18 by holta