iiab/iiab #1139: Admin Console repeatedly failing to install on Raspbian Lite (BIG-sized IIAB)

holta, 2018-09-18:
Re-running `./install` in /opt/iiab/iiab-admin-console did not work &mdash; this error arose 3 times in a row:

> TASK [cmdsrv : Install packages for cmdsrv] ************************************
changed: [127.0.0.1] => (item=libzmq3-dev)
changed: [127.0.0.1] => (item=python-cracklib)
ok: [127.0.0.1] => (item=pkg-config)
failed: [127.0.0.1] (item=sqlite3) => {"cache_update_time": 1537294421, "cache_updated": false, "changed": false, "item": "sqlite3", "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'sqlite3'' failed: E: Unable to correct problems, you have held broken packages.\n", "rc": 100, "stderr": "E: Unable to correct problems, you have held broken packages.\n", "stderr_lines": ["E: Unable to correct problems, you have held broken packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n sqlite3 : Depends: libsqlite3-0 (= 3.16.2-5+deb9u1) but 3.24.0-1 is to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " sqlite3 : Depends: libsqlite3-0 (= 3.16.2-5+deb9u1) but 3.24.0-1 is to be installed"]}
ok: [127.0.0.1] => (item=libapache2-mod-wsgi)
changed: [127.0.0.1] => (item=python-systemd)

Any other ideas?

(This affects IIAB 6.6 and 6.7 equally, as there are essentially no differences in this regard, but FYI this was a BIG-sized installation of IIAB 6.7/master using http://download.iiab.io/6.7/install.txt on RPi 3 B+)

tim-moody, 2018-09-18:
https://github.com/iiab/iiab/blob/master/scripts/calibre-install-pinned-rpi.sh pulls in a test version which breaks Admin Console  install

holta, 2018-09-18:
> https://github.com/iiab/iiab/blob/master/scripts/calibre-install-pinned-rpi.sh pulls in a test version which breaks Admin Console install

FYI/FWIW this did not happen during extensive testing of Calibre 3.30 and 3.31 (e.g. in MEDIUM-sized and BIG-sized installs of IIAB) prior to release of IIAB 6.6 on 2018-09-12.

holta, 2018-09-18:
@jvonau & All: 2 OS's ([Ubuntu 18.10/Cosmic](https://packages.ubuntu.com/cosmic/libsqlite3-0) & [Debian 10/Buster](https://packages.debian.org/buster/libsqlite3-0)) are moving to **libsqlite3-0** 3.24.0-1 but a 3rd OS (Debian Sid, that @arky & someone else I just ran into use) is apparently moving to 3.25.0-1:
https://packages.debian.org/search?keywords=libsqlite3-0

A 4th OS (Ubuntu 18.04 LTS) uses 3.22.0-1:
https://packages.ubuntu.com/search?keywords=libsqlite3-0

Jerry: any recommendations to helps us navigate this swamp, reducing the worst risks over the coming year &mdash; as Raspbian too eventually dumps 3.16.2-5+deb9u1 ?

tim-moody, 2018-09-18:
add apt install sqlite3 to the pinned-rpi script

holta, 2018-09-18:
> add apt install sqlite3 to the pinned-rpi script

On which line best?

CC @floydianslips

holta, 2018-09-19:
> > add apt install sqlite3 to the pinned-rpi script
> 
> On which line best?

"think either before or after `apt -y install /opt/iiab/downloads/calibre*.deb`"

(But this proposed solution is still being evaluated...@floydianslips believes it works but @tim-moody is not sure.)

tim-moody, 2018-09-19:
Just retested on rpi 3:
apt update
apt -y install sqlite3
rm /etc/apt/sources.list.d/rpi-testing.list
apt update

and ran iiab-cmdsrv standalone.
saw no errors
ran iiab-cmdsrv-ctl TEST iiab-cmdsrv-ctl GET-JOB-STAT
without error

also started the console app and init finished without errors

Possibly the error I had before came from not rerunning apt update after removing the test source.

tim-moody, 2018-09-19:
So, although inelegant, this approach seems to work.

holta, 2018-09-19:
sqlite3 is not included out-of-the-box in Raspbian Lite:
```
pi@raspberrypi:~ $ apt -a list sqlite3
Listing... Done
sqlite3/stable 3.16.2-5+deb9u1 armhf
```
What other IIAB service/role/stage is installing sqlite3 prior to Calibre & Admin Console?

(Or might need sqlite3 in future?)

PS interim fix for #1139 is PR #1140 (so Calibre & Admin Console can both install) is now merged.  This will be backported to IIAB 6.6 later if it proves solid in broader testing.

holta, 2018-09-19:
> What other IIAB service/role/stage is installing sqlite3 prior to Calibre & Admin Console?

Answer: nothing is installing sqlite3 until Calibe in Stage 9 and Admin Console after that.  This was reconfirmed by carefully watching a BIG-sized installed of 6.7/master on Rasp Lite on RPi 3 B+.

Closing for now, as this seems resolved, and if so a backport to IIAB 6.6 will be needed soon.

closed 2018-09-19 by holta

holta, 2018-09-19:
FYI this was successfully tested on a BIG-sized install of http://download.iiab.io/6.7/install.txt on Raspbian Lite on RPi 3 B+ and also backported to IIAB 6.6 via PR #1145.

jvonau, 2018-09-19:
There looks to be some issues with the naming of php{{ php_version }}-sqlite3 where there are conditionals based upon distro in https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/main.yml. Think the intent was to have php's sqlite3 connector, that I would think might pull sqlite's server package in, pre-installed for admin-console which is expecting a complete LAMP stack to be already installed (historically this was done in https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/yum-historical.yml). Think the best place to install sqlite3 would be https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/main.yml. as originally intended and not the workaround that is now in place.

holta, 2018-09-19:
> Think the best place to install sqlite3 would be https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/main.yml

If an implementer wants neither Calibre nor Admin Console (it happens, in both cases) what is the overhead they'd be carrying by installing sqlite3 ? (from raspbian/testing or raspbian/stable or whatever)

commit ed8e1571 (2018-09-19): sqlite3 patch: same as #1140, to fix #1139
commit 6653593f (2018-09-19): Merge pull request #1145 from iiab/6.6-patch

Backport of PR #1140 for IIAB 6.6, to fix #1139 sqlite3 so Calibre & Admin Console both install