iiab/iiab #1141: OpenVPN starts too late in IIAB's install--remote support cannot wait 30-45 min

holta, 2018-09-19:
I'd like @jvonau & others to help refine where OpenVPN can be installed+enabled _far earlier_ than [Stage 4](https://github.com/iiab/iiab/blob/master/roles/4-server-options/tasks/main.yml#L13-L17) (as close as possible to the very beginning of IIAB's installation process, to make remote support realistic!)

jvonau, 2018-09-19:
There is an interactive question and early activation in https://github.com/iiab/iiab-factory/pull/33/files

holta, 2018-09-19:
I generally lean against having an interactive question for OpenVPN support, as I think this opens the door to too many people using a collegial service (whether their intentions sincere or not, and whether their security practices are secure or not ;)

Folks who wants service/support should generally talk to us and explain themselves first in my opinion.  Eventually we can send them to a paid support vendor of one kind or another if we cannot handle the volume (not an unlikely scenario!)

Anyway this is the reason I removed OpenVPN support for 1-line installers for the general public from IIAB 6.6, as I'm concerned the risks are too high, with so many important contributors homes being put at potential risk as we now begin to grow.

holta, 2018-09-19:
Closing as PR #1148 merged after successful testing.

(Feel free to re-open this issue if a larger discussion is useful!)

closed 2018-09-19 by holta