iiab/iiab #1142: "Start and enable postgresql-iiab service" fails intermittently on RPi installs

holta, 2018-09-19:
Re-running "curl d.iiab.io/6.7/install.txt | sudo bash" worked.

> TASK [postgresql : Start and enable postgresql-iiab service] *******************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service postgresql-iiab: Job for postgresql-iiab.service failed because the control process exited with error code.\nSee \"systemctl status postgresql-iiab.service\" and \"journalctl -xe\" for details.\n"}

But this is not the 1st time this has happened...is this a timing issue we might address?

tim-moody, 2018-09-19:
what did systemctl status postgresql-iiab.service\" and \"journalctl -xe\" say? it used to happen that stock PostgreSQL was started and then stopped in order to start a custom config, but the custom config failed because the stock psql had the port.

holta, 2018-12-10:
Highly infrequent issue so am closing.

closed 2018-12-10 by holta