iiab/iiab #1143: De-emphasize roles/osm as few people use it anymore

holta, 2018-09-19:
roles/osm takes up ~5min during IIAB installation on RPi and basically nobody uses it anymore.

It should be dramatically de-emphasized particularly since OER2GO/RACHEL module & George's http://download.iiab.io/content/OSM/vector-tiles/ are both now far more compact.

Of course this 1 line is still needed:
```
Alias /maps /library/www/html/modules/en-osm-omt-min/
```
In /etc/apache2/sites-enabled/osm.conf -> /etc/apache2/sites-available/osm.conf

Currently sourced from: https://github.com/iiab/iiab/blob/master/roles/osm/templates/osm.conf.j2#L12

closed 2018-09-20 by holta

tim-moody, 2018-09-20:
of course it remains the only way to have 16 levels of zoom for entire planet

holta, 2018-09-20:
> of course it remains the only way to have 16 levels of zoom for entire planet

Zoom levels 0-to-15 from 2012/2013's tiles might not only be stale & occupy much more disk, but also offer worse resolution than George's current approach which overzooms elegantly to (about) level 18.  Anybody starting new will not be interested in the older 2012/2013 bitmaps / raster maps.

But I've left roles/osm in place just in case one or two diehard implementers want variables {osm_install, osm_enabled} to be set True, to enact this older OSM system.

FYI our new vector-based approach can be slower, and make rendering mistakes on RAM-constrained client devices, so some folk will prefer OER2GO's approach (its National Geographic skin / sex-appeal / shinyness remains a bit more appealing for continent/country-level browsing, than our vector product, for now!)  @georgejhunt & I will try to works on these angles where we can at [OSM's summit in Detroit](https://2018.stateofthemap.us/) in 2 weeks!