iiab/iiab #1146: Is IIAB setting Apache/systemd/etc permissions in a non-standard way?

holta, 2018-09-19:
When installing software on Raspbian after IIAB is installed (e.g. emacs, and others) I commonly get this message:
```
 14 files changed, 2995 insertions(+), 1 deletion(-)
 create mode 120000 apache2/mods-enabled/socache_shmcb.load
 create mode 120000 apache2/mods-enabled/ssl.conf
 create mode 120000 apache2/mods-enabled/ssl.load
 create mode 100644 apache2/sites-available/admin-console.conf
 create mode 100644 apache2/sites-available/cmd-service.conf
 create mode 120000 apache2/sites-enabled/admin-console.conf
 create mode 120000 apache2/sites-enabled/cmd-service.conf
 create mode 100644 iiab/config_vars.yml
 create mode 100644 iiab/kiwix_catalog.json
 create mode 100644 iiab/oer2go_catalog.json
 create mode 100644 systemd/system/iiab-cmdsrv.service
 create mode 120000 systemd/system/multi-user.target.wants/iiab-cmdsrv.service
```
Should we modify chmod permission of any of the above, to better follow standards?

jvonau, 2018-09-19:
With etckeeper creating a git repo of /etc and records the changes when apt runs... Think nobody is really using etckeeper and it's bloating the filesystem every time apt runs.

holta, 2018-09-19:
> With etckeeper creating a git repo of /etc and records the changes when apt runs... Think nobody is really using etckeeper and it's bloating the filesystem every time apt runs.

Ok.  Should etckeeper be commented out from Line 51 of [roles/2-common/tasks/packages.yml](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L51) now?

holta, 2018-09-19:
> Ok. Should etckeeper be commented out from Line 51 of [roles/2-common/tasks/packages.yml](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/packages.yml#L51) now?

Idea added to http://minutes.iiab.io call agenda for Thur 2018-09-20 Item 2. a. so it's not lost or forgotten.

closed 2018-12-10 by holta