iiab/iiab #1147: WordPress (or PHP?) blocks uploading of graphics > 2MB (WP defaults not acceptable?)

holta, 2018-09-19:
@ericnitschke asks if IIAB can consider new defaults to /etc/php/7.0/apache2/php.ini (and similar, across all OS's) to something like this?
```
upload_max_filesize 64M    ; default on RPi is 2M
post_max_size 128M    ; default on RPi is 8M
memory_limit 256M    ; default on RPi is 128M 
max_execution_time 300    ; default on RPi is 30
max_input_time 300    ; default on RPi is 60
```
(At a glance I'm concerned about modifying 5 defaults on such a RAM-constrained platform, and hope @kananigit and others can help investigate this, so occasional WordPress users still have the choice to keep their RPi machines responsive w/o too many resources grabbed by WordPress?)

holta, 2018-09-19:
@ericnitschke also has questions about:
- WordPress plugins being blocked (not installable if they are too large?)
- Wordpress' content export/import functions not fully working.

I've asked him to create separate issues at https://github.com/iiab/iiab/issues so others can understand what exactly he's trying to do in each case &mdash; on exactly which versions of {WordPress, IIAB, underlying OS, etc} so others can fully reproduce this &mdash; which he will do that in coming days.

holta, 2018-09-19:
Please also see:
- https://github.com/iiab/iiab/issues/201 Automate re-homing of http://HOSTNAME/wordpress? (instead of RELOCATE flag)
- https://github.com/iiab/iiab/issues/992 " http://box/wordpress + http://OTHER-DOMAIN/wordpress + http://OTHER-IP/wordpress "

holta, 2018-09-19:
@ericnitschke asks if https://en.wikipedia.org/wiki/BuddyPress works to create a social network / community-of-practice on WordPress on RPi, @kananigit do you know anything about this?

kananigit, 2018-09-19:
Updating the values in /etc/php/7.0/apache2/php.ini   is normal when dealing with platforms like wordpress and Moodle even when working with shared web hosting. You get an error when uploading a file which is bigger than the limit or when trying to export a big file. 

I do that all the time when working on moodle or WP on IIAB servers. You wanna be careful with the memory limits though. I agree with @holta , Since this is dependent on what an implementer is trying to upload , we cannot just have some set default values. We could note the issue somewhere so that people know how to edit them.  

I have a quick and dirty script that i wrote to help me update this values
https://github.com/kananigit/Automation_Scripts/blob/master/update_php_ini.sh

I have also considered having the values somewhere in the admin console so that they could be updated from a web interface since they need to be updated inorder to use WP and Moodle meaningfully.

As for Buddypress social networking i believe you should be able to use it just fine since its just a wordpress plugin. I have not personally tried it though.

holta, 2018-09-19:
Profound thx @kananigit:

Have WordPress's content export/import functions worked for you?  (Eric could not get them to work, as I've asked him to document in detail tomorrow, so others can understand/reproduce the exact issue.)

PS let's try to work with @ericnitschke tmrw to get WordPress best practices unified into http://FAQ.IIAB.IO so any grandpa/grandma (or 10-year-old ;-) can install IIAB/WordPress & get rolling _pronto._

ericnitschke, 2018-09-21:
Thanks @holta and @kananigit. 

> upload_max_filesize 64M    ; default on RPi is 2M
> post_max_size 128M    ; default on RPi is 8M
> memory_limit 256M    ; default on RPi is 128M 
> max_execution_time 300    ; default on RPi is 30
> max_input_time 300    ; default on RPi is 60

The only barriers we've faced so far with the above are the upload and post max filesize and the max execution limit.  The max upload was preventing us from installing plugins and uploading media.  The execution time was causing issues when installing plugin content (for example from https://visualcomposer.io - which has been a good/easy resource for building sites offline).

Can we change these defaults and leave the memory defaults the same? Otherwise I second the idea of "having the values somewhere in the admin console" to make it easier for Wordpressers/Moodlers.

Buddypress makes it quite easy to turn WP in a full-fledged LMS.  I was just curious to hear if you know of anyone using this in offline contexts.

We've made a few pretty awesome landing pages in Wordpress, but have encountered the export/import issue listed above (which we can turn into separate ticket if desired).

We would like to copy Wordpress pages/media/posts/etc/ between other IIAB installs/versions. The native WP export/import (https://www.abrightclearweb.com/export-import-content-wordpress-site/) has failed the last few times.  Will try again soon to make sure the plugins are the exact same before import which might be causing the issue.

Once this is resolved we can start sharing IIAB landing page templates widely for different contexts.  It would be nice to have an IIAB "theme" directory. 

Another issue of immediate importance is having box.lan/wordpress the default page upon typing (http://)box.lan.  There's an option to select this as the server home page in the admin panel, but we haven't been able to get it to work in 6.5.  @holta said this was fixed in 6.6/7 which I'll test this weekend.

More to follow.  Thanks @all for the help.

holta, 2018-09-21:
> We've made a few pretty awesome landing pages in Wordpress, but have encountered the export/import issue listed above (which we can turn into separate ticket if desired)...We would like to copy Wordpress pages/media/posts/etc/ between other IIAB installs/versions.

Yes: please open a separate ticket with all details, so we can solve this promptly.  Thanks Eric!

> immediate importance is having box.lan/wordpress the default page upon typing (http://)box.lan. There's an option to select this as the server home page in the admin panel, but we haven't been able to get it to work in 6.5. @holta said this was fixed in 6.6/7 which I'll test this weekend.

Thanks!  Plz open an issue/tkt here if glitches arise.

holta, 2018-09-21:
@kananigit can you post WordPress settings you're using at institutions in Kenya, so we can consolidate/publish healthy community norms for IIAB implementers & all?

> Buddypress makes it quite easy to turn WP in a full-fledged LMS. I was just curious to hear if you know of anyone using this in offline contexts.

@kananigit can you check this out?  Or does this not fit into your plans in Kenya?

kananigit, 2018-09-21:
@holta  I think the defaults Eric has above will work fine. Its about what i have used personally. I have only had to change 
upload_max_filesize 64M ; default on RPi is 2M
post_max_size 128M ; default on RPi is 8M

I know of buddypress but haven't played with it. For LMS am a little inclined to moodle but will checkout buddypress sometime.

@ericnitschke 
On having wordpress as the homepage, i suggest editing local vars when doing a fresh install 
iiab_home_url: /home     --->    iiab_home_url: /wordpress

especially if you only use wordpress on your installations. I have had success with that on my tests. Generally its great to have your customized version of local_vars.

If you could also share some of the exported data you are  trying to move to other iiab servers and detail the versions and what exactly is going wrong so we can try to replicate. You could do it on a different ticket as you mentioned above.

Am also interested to know whether you are, or are planning to use wordpress from IIAB as a standalone server or as part of an existing network or both.

holta, 2018-09-22:
Please see PR #1161 "wordpress_raise_php_limits variable for schools that use Wordpress intensively" that may not be the final word, but is tested on Raspbian and Ubuntu 18.04 to deliver what @ericnitschke and @kananigit proposed above.

_Thank you for providing feedback in coming days after [testing it](https://github.com/iiab/iiab/pull/1161#issuecomment-423752128) on [IIAB 6.7](http://download.iiab.io/6.7/) (master)._

holta, 2018-09-22:
@jvonau & All I'm started to wonder if /etc/php.ini (SEE #69) might be largely or entirely ignored in 2018 ??

Raspbian's http://box/info.php shows:
```
PHP Version 7.0.30-0+deb9u1
Loaded Configuration File | /etc/php/7.0/apache2/php.ini
```
...and:
```
root@box:~# php -i | grep php.ini
Configuration File (php.ini) Path => /etc/php/7.0/cli
Loaded Configuration File => /etc/php/7.0/cli/php.ini
```
Ubuntu 18.04's http://box/info.php shows:
```
PHP Version 7.2.10-0ubuntu0.18.04.1
Loaded Configuration File | /etc/php/7.2/apache2/php.ini
```
...and:
```
root@box:~# php -i | grep php.ini
Configuration File (php.ini) Path => /etc/php/7.2/cli
Loaded Configuration File => /etc/php/7.2/cli/php.ini
```

jvonau, 2018-09-23:
I feel that /etc/php.ini should just go away and deal with the timezone via some other way.

holta, 2018-09-23:
> I feel that /etc/php.ini should just go away and deal with the timezone via some other way.

Great catch: https://github.com/iiab/iiab/blob/master/roles/httpd/templates/php.ini.j2#L636

I'll revise PR #1161 to do this, and refocus our efforts on the httpd (Apache) playbook instead of WordPress playbook.  Within ~36h when I get a break from traveling.

holta, 2018-09-24:
@kananigit and @ericnitschke: the variable has been renamed from `wordpress_raise_php_limits` to `apache_raise_php_limits` and is now a part of [IIAB 6.7](http://download.iiab.io/6.7/) / master, as a result of PR #1165.

(Please check it out by setting this variable to True on [Line 84](https://github.com/iiab/iiab/blob/master/vars/local_vars_medium.yml#L84) of [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) before installing!)

holta, 2018-09-24:
Please see #1166 "Rename var apache_raise_php_limits to apache_high_php_limits [for WordPress & Moodle]" which is now part of [IIAB 6.7](http://download.iiab.io/6.7/) / master, thanks all.

holta, 2018-09-28:
@ericnitschke & @kananigit after 9 days it appears IIAB is solid in its WordPress support.  If remaining or new issues arise, please open a new ticket, Thanks!

closed 2018-09-28 by holta

holta, 2018-11-07:
FYI @ericnitschke has made WordPress progress here and will shortly suggest:
- new/higher defaults when `apache_raise_php_limits: True` ...going above and beyond raising `upload_max_filesize = 2M` to `upload_max_filesize = 64M` (perhaps up to 128M or 200M or so?)
- how we can now export/import WordPress landing/splash pages, or entire sites/courses between IIAB versions, without having to install plugins and re-align images, etc?

I and others will then document community norms/recommendations in places like:
http://FAQ.IIAB.IO
http://wiki.laptop.org/go/IIAB/FAQ#Can_I_optimize_WordPress_and_Moodle_for_high-traffic_usage.3F