iiab/iiab #1158: "Unable to start service kolibri: Job for kolibri.service failed because a timeout was exceeded."

holta, 2018-09-20:
@arky do you have ideas on solving this, near the very end of the kolibri playbook?  It seems to happens 20% of the time (roughly?) when installing IIAB 6.6 or [IIAB 6.7](http://download.iiab.io/6.7/) (master) on Raspberry Pi:

> TASK [kolibri : Enable & (Re)Start kolibri service] ****************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service kolibri: Job for kolibri.service failed because a timeout was exceeded.\nSee \"systemctl status kolibri.service\" and \"journalctl -xe\" for details.\n"}

FYI running those 2 commands about ("systemctl status kolibri.service" and "journalctl -xe") gives no further info unfortunately.

arky, 2018-09-21:
@holta  This is a critical problem. Please kindly assign this to me. I will work on getting this resolved this weekend.

holta, 2018-09-21:
Is this a duplicate of #1094 ?

arky, 2018-09-21:
@holta Not quite sure, Perhaps I need to replicate/debug the problem to better understand that.

holta, 2018-09-22:
> @holta Not quite sure, Perhaps I need to replicate/debug the problem to better understand that.

I believe #1094 is the same issue.  It happens often enough to be a real pain.  Not on every RPi install for sure, but definitely it's common enough to be a real hassle seeding confusion among new implementers.

(FYI it's happened to me at least 3 times over the past 2 weeks, which is more than 10% of the time FWIW!)

arky, 2018-09-23:
@holta Kolibri upstream developers and I are actively working on solution for this problem. A stop gap patch is on the way.  :)

arky, 2018-09-26:
@holta Let me know if  #1164 resolves the problem. If so, you can close this bug as resolved. Thanks

holta, 2018-09-26:
> @holta Let me know if #1164 resolves the problem. If so, you can close this bug as resolved. Thanks

Sounds extremely promising:

Let's close this tkt if the problem (#1158 "Unable to start service kolibri: Job for kolibri.service failed because a timeout was exceeded.") doesn't recur over the coming month or so.

arky, 2018-09-26:
@holta Sounds good. 

Meanwhile I recommend following up similar any problems in the following upstream bug. There is very little we can do to resolve them here in downstream IIAB implementation of IIAB.  

https://github.com/learningequality/kolibri/issues/4310

closed 2018-12-10 by holta