iiab/iiab #1163: KA Lite failed to pip install on Raspbian Liteon RPi 3 B+

holta, 2018-09-22:
@jvonau is there anything to learn from this?  When I reran ./iiab-install it completed successfully the 2nd time:

> TASK [kalite : Install KA Lite static with pip (debuntu)] **********************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/bin/virtualenv --no-site-packages /usr/local/kalite/venv", "msg": "stdout: New python executable in /usr/local/kalite/venv/bin/python2\nAlso creating executable in /usr/local/kalite/venv/bin/python\nInstalling setuptools, pkg_resources, pip, wheel...\n  Complete output from command /usr/local/kalite/venv/bin/python2 - setuptools pkg_resources pip wheel:\n  Collecting setuptools\nException:\nTraceback (most recent call last):\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/basecommand.py\", line 215, in main\n    status = self.run(options, args)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/commands/install.py\", line 353, in run\n    wb.build(autobuilding=True)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/wheel.py\", line 749, in build\n    self.requirement_set.prepare_files(self.finder)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_set.py\", line 380, in prepare_files\n    ignore_dependencies=self.ignore_dependencies))\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_set.py\", line 554, in _prepare_file\n    require_hashes\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_install.py\", line 278, in populate_link\n    self.link = finder.find_requirement(self, upgrade)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py\", line 465, in find_requirement\n    all_candidates = self.find_all_candidates(req.name)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py\", line 423, in find_all_candidates\n    for page in self._get_pages(url_locations, project_name):\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py\", line 568, in _get_pages\n    page = self._get_page(location)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py\", line 683, in _get_page\n    return HTMLPage.get_page(link, session=self.session)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py\", line 792, in get_page\n    \"Cache-Control\": \"max-age=600\",\n  File \"/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 501, in get\n    return self.request('GET', url, **kwargs)\n  File \"/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/download.py\", line 386, in request\n    return super(PipSession, self).request(method, url, *args, **kwargs)\n  File \"/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 488, in request\n    resp = self.send(prep, **send_kwargs)\n  File \"/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py\", line 609, in send\n    r = adapter.send(request, **kwargs)\n  File \"/usr/local/kalite/venv/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py\", line 47, in send\n    resp = super(CacheControlAdapter, self).send(request, **kw)\n  File \"/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py\", line 423, in send\n    timeout=timeout\n  File \"/usr/local/kalite/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py\", line 643, in urlopen\n    _stacktrace=sys.exc_info()[2])\n  File \"/usr/local/kalite/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py\", line 315, in increment\n    total -= 1\nTypeError: unsupported operand type(s) for -=: 'Retry' and 'int'\n----------------------------------------\n...Installing setuptools, pkg_resources, pip, wheel...done.\nRunning virtualenv with interpreter /usr/bin/python2\n\n:stderr: Traceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 2375, in <module>\n    main()\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 724, in main\n    symlink=options.symlink)\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 992, in create_environment\n    download=download,\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 922, in install_wheel\n    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 817, in call_subprocess\n    % (cmd_desc, proc.returncode))\nOSError: Command /usr/local/kalite/venv/bin/python2 - setuptools pkg_resources pip wheel failed with error code 2\n"}

jvonau, 2018-09-22:
Cleaned up the traceback for readabliity fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/bin/virtualenv --no-site-packages /usr/local/kalite/venv", "msg": "stdout: New python executable in /usr/local/kalite/venv/bin/python2\n
Also creating executable in /usr/local/kalite/venv/bin/python\n
Installing setuptools, pkg_resources, pip, wheel...\n 
Complete output from command /usr/local/kalite/venv/bin/python2 - setuptools pkg_resources pip wheel:\n 
Collecting setuptools\n
Exception:\n
Traceback (most recent call last):\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/basecommand.py", line 215, in main\n 
status = self.run(options, args)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/commands/install.py", line 353, in run\n 
wb.build(autobuilding=True)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/wheel.py", line 749, in build\n 
self.requirement_set.prepare_files(self.finder)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_set.py", line 380, in prepare_files\n 
ignore_dependencies=self.ignore_dependencies))\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_set.py", line 554, in _prepare_file\n 
require_hashes\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/req/req_install.py", line 278, in populate_link\n 
self.link = finder.find_requirement(self, upgrade)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 465, in find_requirement\n 
all_candidates = self.find_all_candidates(req.name)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 423, in find_all_candidates\n 
for page in self._get_pages(url_locations, project_name):\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 568, in _get_pages\n 
page = self._get_page(location)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 683, in _get_page\n 
return HTMLPage.get_page(link, session=self.session)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/index.py", line 792, in get_page\n 
"Cache-Control": "max-age=600",\n 
File "/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get\n 
return self.request('GET', url, **kwargs)\n 
File "/usr/share/python-wheels/pip-9.0.1-py2.py3-none-any.whl/pip/download.py", line 386, in request\n 
return super(PipSession, self).request(method, url, *args, **kwargs)\n 
File "/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request\n 
resp = self.send(prep, **send_kwargs)\n 
File "/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send\n 
r = adapter.send(request, **kwargs)\n 
File "/usr/local/kalite/venv/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send\n 
resp = super(CacheControlAdapter, self).send(request, **kw)\n 
File "/usr/local/kalite/venv/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send\n 
timeout=timeout\n 
File "/usr/local/kalite/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen\n 
_stacktrace=sys.exc_info()[2])\n 
File "/usr/local/kalite/venv/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment\n 
total -= 1\n
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'\n
----------------------------------------\n
...Installing setuptools, pkg_resources, pip, wheel...done.\n
Running virtualenv with interpreter /usr/bin/python2\n
\n
:stderr: Traceback (most recent call last):\n 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 2375, in \n 
main()\n 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main\n symlink=options.symlink)\n 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 992, in create_environment\n download=download,\n 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 922, in install_wheel\n call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)\n 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess\n % (cmd_desc, proc.returncode))\n
OSError: Command /usr/local/kalite/venv/bin/python2 - setuptools pkg_resources pip wheel failed with error code 2\n"}

holta, 2018-11-04:
I haven't seen this in 2 months, so let's declare victory for now!

closed 2018-11-04 by holta