iiab/iiab #1170: Vector Map of Haiti + DR + S. Florida + Greater NYC + Boston + Montreal (+ Brazil?)

holta, 2018-09-26:
There is a need for disconnected Haitian people to visualize where their family members have gone.

This should be accomplished by an .mbtiles file that compactly covers most/all of these 7 disparate regions (listed above, in subject line) where large numbers of Haitian and Haitian diaspora people live!

Even if street name search is not yet possible at this time, a compact such .mbtiles file remains a critical need — i.e. one that does NOT include all of Canada, USA, Central America and South America — _ideally much closer to 2GB than the 20GB that all of N/S America would occupy!_

I will try to make such happen with @georgejhunt's help and his ideas, who I'm meeting at OpenStreetMap's conference (https://2018.stateofthemap.us) in Detroit in early October.

holta, 2018-09-26:
Also all of New Jersey might well be advisable (179 MB).  Connecticut if nec (66 MB).  Or possibly this NYC bounding box (101 MB, which includes nearly half of NJ's population?) might be sufficient: https://openmaptiles.com/downloads/north-america/us/new-york/new-york/

OR the entire Northeast US (1.36 GB) incl Southern Quebec if truly nec??  All of MA is a bit overweight (213 MB, includes CT, RI).  But possibly nec as "Boston" (32 MB) doesn't include any suburbs at all :(  Or, NY State (745 MB) might/maybe be a compromise, as it includes most of MA/NJ/etc except for Cape Cod and southern NJ: https://openmaptiles.com/downloads/dataset/osm/north-america/us/new-york/

Greater Montreal should be included one way or another (70 MB).  Turns out Sherbrooke (23 MB) can be added if there are a lot of Haitians too??  The province of Quebec cannot be added as it has too many rivers etc, and takes too much disk (SD card) space, and strangely is missing from openmaptiles.com entirely (upstream bug!) even if it was wanted.

With all of Florida (244 MB) likely added too.  Atlanta maybe (117 MB) if enough Haitians live there?  All of Brazil (2.11GB) if absolutely nec ??  But if Haitians are concentrated in particular parts of Brazil I need to figure that out 🔍

Interesting how this emerging Haiti Diaspora Map represents a kind of "transnational family" with far more financial wealth and influence than Haiti per se...

(FYI/Likewise the official Haiti+DR are only 89 MB as there are fewer roads/rivers/features on the Hispaniolan Isle itself).

holta, 2018-09-26:
Oops Cuba needs to be included too (98 MB) as it contains something like 300,000+ longstanding Haitian immigrants (also Haitian Creole speakers, who may not be Haitian biologically) according to https://en.wikipedia.org/wiki/Haitian_diaspora

Atlanta (117 MB) and Philadelphia (55 MB) too if they fit, as these are emerging Haitian-American populations of 10,000+

For now I think we can let slide European cities, as Haitians are a bit too spread out over Paris, London, etc.

_Which parts of Brazil to include (or not to include) remains a difficult decision, as is made clear by the above Wikipedia article!_

holta, 2018-09-26:
Oops I totally missed Chile ("105,000 Haitians") and Bahamas ("80,000 Haitians").

(The above 2 are very real, according to Wikipedia & others!  So...)

The problem is that Chile's bounding box includes much of Argentina, and so occupies an enormous 860 MB, most of which is 100% irrelevant to Haitians.  Including Chile's capital alone (Santiago, 38 MB) might be a likely compromise?

Bahamas faces a similar quandary.  The only practical bounding box including the Bahamas appears to be "Central America & Caribbean" weighing in at a heavyweight 1.33 GB :/

Conclusion: to really cut this Haitian Diaspora Map Pack down to size (to do this job properly for use on small microSD cards etc) we should likely focus on "Haitian Cities Worldwide" and occasionally (as in the case of the Bahamas) we MIGHT possibly be forced to create our own bounding boxes (to avoid vacuuming up dozens of irrelevant countries/territories!)

holta, 2018-09-26:
Likewise to avoid the [Chile](https://openmaptiles.com/downloads/dataset/osm/south-america/chile/)'s 860MB tileset (quite unnec in this case) we should just include its large capital [Santiago](https://openmaptiles.com/downloads/dataset/osm/south-america/chile/santiago/) (38 MB) for starters, which is probably sufficient for now.

Of course not all the ~105,000 Haitians in Chile live in the capital, so later if truly nec (and there is room!) we might add ~3 other most likely cities:
- [Valparaiso](https://openmaptiles.com/downloads/dataset/osm/south-america/chile/valparaiso/) (26 MB, just W of capital)
- [Rancagua](https://openmaptiles.com/downloads/dataset/osm/south-america/chile/rancagua/) (23 MB, just S of capital)
- [Concepcion](https://openmaptiles.com/downloads/dataset/osm/south-america/chile/concepcion/) (25 MB, far S of capital)

holta, 2018-09-27:
@deldesir: to reduce the overwhelmingly large size of [Brazil](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/)'s mbtiles data (1.79 - 2.11GB, as Brazil is almost as big as the China, the US and Canada) I'm looking to find which cities in Brazil contain a large number of Haitians.

São Paulo is certainly the hub of Brazil's influx of Haitians since 2010's earthquake (well over 100,000 people, or more than 1% of Haiti's population) so we will definitely include that...but the question boils down to: _are their other Brazilian cities with large Haitian populations...in the list below list especially?_

1. [Sao Paolo](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/sao-paulo/) 49.05 - 64.32 MB
2. [Porto Alegre](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/porto-alegre/) 34.30 -45.95 MB
3. [Rio de Janeiro](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/rio-de-janeiro/) 29.98 - 41.13 MB
4. [Brasilia](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/brasilia/) 20.45 - 30.51 MB
5. [Curitiba](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/curitiba/) 19.83 - 29.50 MB
6. [Belo Horizonte](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/belo-horizonte/) 16.15 - 25.69 MB
7. [Salvador](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/salvador/) 16.70 - 26.37 MB
8. [Florianopolis](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/florianopolis/) 15.89 - 25.10 MB
9. [Recife](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/recife/) 15.60 - 24.88 MB
10. [Fortaleza](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/fortaleza/) 15.40 - 24.61 MB
11. [Blumenau](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/blumenau/) 14.89 - 24.06 MB
12. [Manaus](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/manaus/) 14.59 - 23.52 MB
13. [Campo Grande](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/campo-grande/) 14.57 - 23.43 MB
14. [Teresina](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/teresina/) 13.67 - 22.66 MB
15. [Palmas](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/palmas/) 13.30 - 22.11 MB
16. [Mariana](https://openmaptiles.com/downloads/dataset/osm/south-america/brazil/mariana/) 12.63 - 21.38 MB

aidan-fitz, 2018-10-09:
I think I understand now. We want to add map data for the cities themselves, not information on Haitian households that live there.

holta, 2018-10-29:
@georgejhunt is making steady progress on the 'additive' side of this problem — i.e. merging 2 or more .mbtiles files without excess bloat:
https://github.com/iiab/iiab-factory/pull/52

The 'subtractive' side of this challenge (sunsetting a specific bounding box or any other region from the global .mbtiles vector dataset, also critical) will have to wait a bit longer...

holta, 2018-11-15:
Refs: https://github.com/iiab/iiab-factory/pull/52/ https://github.com/iiab/iiab-factory/pull/53/