iiab/iiab #1183: "./runrole cups" & "./iiab-install --reinstall" repeatedly fail on RPi

holta, 2018-10-08:
@georgejhunt can you take a look at 10.8.0.34 to see if this is easily fixed?

Problem happens every time on this Raspbian Lite on RPi 3 B+ (the original went flawlessly 5 days ago using "MIN-sized" install with very minor modifications &mdash; note that Ansible fails to complete regardless whether one runs Ansible 2.6.5 or 2.7.0).

> TASK [cups : Get the CUPS package installed] ***********************************
[DEPRECATION WARNING]: Invoking "package" only once while using a loop via
squash_actions is deprecated. Instead of using a loop to supply multiple items
and specifying `name: {{ item }}`, please use `name: [u'cups']` and remove the
loop. This feature will be removed in version 2.11. Deprecation warnings can be
 disabled by setting deprecation_warnings=False in ansible.cfg.
failed: [127.0.0.1] (item=[u'cups']) => {"cache_update_time": 1539017377, "cache_updated": false, "changed": false, "item": ["cups"], "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'cups'' failed: E: Unable to correct problems, you have held broken packages.\n", "rc": 100, "stderr": "E: Unable to correct problems, you have held broken packages.\n", "stderr_lines": ["E: Unable to correct problems, you have held broken packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n cups : Depends: libcups2 (= 2.2.1-8+deb9u2) but 2.2.8-5 is to be installed\n        Depends: libcupscgi1 (>= 1.4.2) but it is not going to be installed\n        Depends: libcupsmime1 (>= 1.4.0) but it is not going to be installed\n        Depends: libcupsppdc1 (>= 1.4.0) but it is not going to be installed\n        Depends: cups-core-drivers (>= 2.2.1-8+deb9u2) but it is not going to be installed\n        Depends: cups-daemon (>= 2.2.1-8+deb9u2) but it is not going to be installed\n        Depends: cups-client (>= 2.2.1-8+deb9u2) but it is not going to be installed\n        Depends: cups-ppdc but it is not going to be installed\n        Recommends: colord but it is not going to be installed\n        Recommends: printer-driver-gutenprint but it is not going to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " cups : Depends: libcups2 (= 2.2.1-8+deb9u2) but 2.2.8-5 is to be installed", "        Depends: libcupscgi1 (>= 1.4.2) but it is not going to be installed", "        Depends: libcupsmime1 (>= 1.4.0) but it is not going to be installed", "        Depends: libcupsppdc1 (>= 1.4.0) but it is not going to be installed", "        Depends: cups-core-drivers (>= 2.2.1-8+deb9u2) but it is not going to be installed", "        Depends: cups-daemon (>= 2.2.1-8+deb9u2) but it is not going to be installed", "        Depends: cups-client (>= 2.2.1-8+deb9u2) but it is not going to be installed", "        Depends: cups-ppdc but it is not going to be installed", "        Recommends: colord but it is not going to be installed", "        Recommends: printer-driver-gutenprint but it is not going to be installed"]}

holta, 2018-10-09:
(Clarif: rebooting does not resolve this.)

georgejhunt, 2018-10-09:
This appears to be a dependency problem:
root@box:/etc/apt/sources.list.d# apt-get install cups
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 cups : Depends: libcups2 (= 2.2.1-8+deb9u2) but 2.2.8-5 is to be installed
        Depends: libcupscgi1 (>= 1.4.2) but it is not going to be installed
        Depends: libcupsmime1 (>= 1.4.0) but it is not going to be installed
        Depends: libcupsppdc1 (>= 1.4.0) but it is not going to be installed
        Depends: cups-core-drivers (>= 2.2.1-8+deb9u2) but it is not going
to be installed
        Depends: cups-daemon (>= 2.2.1-8+deb9u2) but it is not going to be
installed
        Depends: cups-client (>= 2.2.1-8+deb9u2) but it is not going to be
installed
        Depends: cups-ppdc but it is not going to be installed
        Recommends: colord but it is not going to be installed
        Recommends: printer-driver-gutenprint but it is not going to be
installed
E: Unable to correct problems, you have held broken packages.
root@box:/etc/apt/sources.list.d#

Is this on top of a stale install.  I suspect that a clean install will
just work. The googling of this error message talks about removing
libecups2, and reinstalling, which we cannot really do in a playbook.

holta, 2018-10-09:
> Is this on top of a stale install.

No, the minimal install was 5 days prior to yesterday, and it was not touched since.

I'd like to understand how this failure happened, so others are not blocked from running `./iiab-install --reinstall` and Admin Console.

If @georgejhunt or @jvonau can see how this happened on 10.8.0.34 ?

holta, 2018-10-09:
FWIW reverting libgs9-common to 9.20~dfsg-3.2+deb9u5 solved the problem &mdash; the root cause / problemati version 9.25 appears here:

> root@box:/opt/iiab/iiab# apt -a list libgs9-common
Listing... Done
libgs9-common/now 9.25~dfsg-2 all [installed,local]
libgs9-common/stable 9.20~dfsg-3.2+deb9u5 all

( Very unclear what part of IIAB caused the installation of 9.25, triggering this problem, but at least now we can keep an eye on it ! )

holta, 2018-10-12:
@floydianslips also faced this CUPS installation problem...on a fresh install on the new Raspbian Lite 2018-10-09.

I'm not able to reproduce it, so there's definitely something intermittent (and irritating) happening here that perhaps it's time to change CUPS's defaults to no longer install unless absolutely necessary.

holta, 2018-12-10:
Closing as this has not been an issue for 2+ months.

closed 2018-12-10 by holta