iiab/iiab #1190: Gitea, GitLab, GitHub Classroom

aidan-fitz, 2018-10-09:
We should consider adding one of these to the IIAB project.

- [Gitea](https://gitea.io/en-us/) is a lightweight Git server written in Go that can run on Raspberry Pi. **I recommend Gitea for the IIAB project.**
- [GitLab Community Edition](https://gitlab.com/gitlab-org/gitlab-ce) is an open-core Git server written in Ruby. Recommended 2 cores and 8 GB of RAM (see [full requirements](https://gitlab.com/gitlab-org/gitlab-ce/blob/master/doc/install/requirements.md)).
- [GitHub Classroom](https://github.com/education/classroom) is explicitly designed for classroom use, but might require teachers and/or students to have GitHub accounts.

Packaging a Git server for use on IIAB would help enable schools to teach CS and info sci classes. It would also enable all institutions to share code that [automates work-related tasks](https://automatetheboringstuff.com/). In both cases, we should have to provide a standard suite of development tools.

aidan-fitz, 2018-10-09:
Gitea can be installed from source or from binary.

- [Install from source](https://docs.gitea.io/en-us/install-from-source/) (requires Go)
- [Install from binary](https://docs.gitea.io/en-us/install-from-binary/)

I think we should use the binary one, but it requires that we continually update the version number in the Ansible config. We could ask the maintainers to create a "latest stable" shortcut.

The Gitea maintainers provide a `systemd` script [here](https://docs.gitea.io/en-us/linux-service/).

arky, 2018-10-09:
@aidan-fitz Interesting projects, Few years ago I had traveled with github team teaching git across East Africa. Lot of changed since then,  Will check out these projects. Thanks!

aidan-fitz, 2018-10-18:
I sent out a message to the Gitea discord asking if anyone wants to work with me on this - let's see what happens!

coolaj86, 2018-10-18:
I have a script that downloads the latest version, creates users, installs systemd, etc if you'd like to use it as a template:

https://git.coolaj86.com/coolaj86/gitea-installer.sh

PRs welcome if it doesn't do what you need and you think your need is general enough to include. :)

tolidano, 2018-10-24:
Cool idea, a couple of questions:
- Why Gitea over Gogs itself?
- Would Gerrit + Gitolite/Gitweb do the same job?

techknowlogick, 2018-10-24:
@tolidano 

> Why Gitea over Gogs itself?

https://docs.gitea.io/en-us/comparison/ and https://blog.gitea.io/2016/12/welcome-to-gitea/

(Full disclosure: I am one of the maintainers of Gitea)

m-anish, 2019-03-17:
@aidan-fitz @holta can this be closed?

closed 2019-03-17 by aidan-fitz

aidan-fitz, 2019-03-17:
> Would Gerrit + Gitolite/Gitweb do the same job?

@tolidano I don't want this question to fall by the wayside, so I'll respond to the best of my ability. I don't know what Gerrit's system requirements or memory and CPU footprint are. [Google's documentation][install gerrit] says:

> 1. A Unix-based server, including any Linux flavor, MacOS, or Berkeley Software Distribution (BSD).
> 2. Java SE Runtime Environment version 1.8. Gerrit is not compatible with Java 9 or newer yet.

Resource efficiency is paramount for us because the software needs to be runnable on Raspberry Pi&ndash;based devices.

> Why Gitea over Gogs itself?

What @techknowlogick said. But are there features that Gogs has but Gitea doesn't?

[install gerrit]: https://gerrit-documentation.storage.googleapis.com/Documentation/2.16.7/linux-quickstart.html