iiab/iiab #1191: sqlite3 installation under 3-base-server (should this be mandatory?)

holta, 2018-10-09:
@georgejhunt indicates he may work on this incoming weeks.

Should this appear right next to SQLite presumably, around Line 17 of https://github.com/iiab/iiab/blob/master/roles/3-base-server/tasks/main.yml#L17 ?

For those implementers who do not need Captive Portal and Calibre, should this be mandatory?

Ref PRs: #1179 #1180 #1185 #1187

holta, 2018-10-09:
@georgejhunt feel free to un-assign yourself (if you no longer want to work on this!)

holta, 2018-10-09:
@georgejhunt & All, please see related ticket #1206

tim-moody, 2018-10-09:
if it is part of base it is mandatory. otherwise it remains a dependency of several roles

holta, 2018-10-14:
Is @georgejhunt's PR #809 "load php-sqlite on ubuntu" partially relevant, from May 2018?

holta, 2019-12-26:
@jvonau & All: can we close this tkt?

(Or act on it, as nec?)

jvonau, 2019-12-26:
Once #2070 with the addition of sqlite3 in 2-common's packages.yml lands this issue would be resolved.

holta, 2020-01-02:
sqlite3 is not being installed by 2-common, as a result of PR #2070

closed 2020-01-02 by holta