iiab/iiab #1194: Admin Console can fail to install on RPi: "Gathering Facts" timer expires in 10 sec

holta, 2018-10-09:
FYI this was the usual BIG-sized install of http://download.iiab.io/6.7 onto Raspbian Lite on RPi 3 B+:

> TASK [Gathering Facts] *********************************************************
An exception occurred during task execution. To see the full traceback, use -vvv. The error was: TimeoutError: Timer expired after 10 seconds
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/bin/findmnt --list --noheadings --notruncate", "msg": "Timer expired after 10 seconds", "rc": 257}
        to retry, use: --limit @/opt/iiab/iiab-admin-console/iiab-admin.retry
>
> PLAY RECAP *********************************************************************
> 127.0.0.1                  : ok=0    changed=0    unreachable=0    failed=1

tim-moody, 2018-10-09:
think this is set by iiab. not sure why it would fail in admin console

tim-moody, 2018-10-16:
actually, I had forgotten that adm cons does add additional facts:

ANSIBLE_PROGRAM=`which ansible`
ANSIBLE__PLAYBOOK_PROGRAM=`which ansible-playbook`

hard for me to imagine that which would timeout unless the device is busy with something else

holta, 2018-10-19:
Recap from today's community/team call @ http://minutes.iiab.io

@tim-moody is aware of the issue, but doesn't have time to work on this at the moment.  Others should feel free to.

Clarif: `cd /opt/iiab/iiab-admin-console` followed by re-running `./install` will almost always work to install Admin Console.

tim-moody, 2018-10-19:
To be clear, I have never seen this issue and cannot reproduce.

tim-moody, 2018-10-19:
possibly related
https://github.com/ansible/ansible/issues/43884

holta, 2018-10-19:
> possibly related
ansible/ansible#43884

Interesting.  Would it help to increase the timeout from 10 seconds to 100 seconds (or even 200 seconds) as iiab/iiab does at [default_vars.yml#L26-L27](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L26-L27) ?

Background: the problem occurred on a BIG-sized install of IIAB on RPi 3 B+, which naturally had a high load average after ./iiab-install was just completing.

holta, 2019-07-26:
@jvonau recommends closing this.  But if this recurs, please re-open.

closed 2019-07-26 by holta